SpikeInterface/spikeinterface #3818: No waveforms displayed when using plot_unit_waveforms after Kilosort4 analysis

cuowunimei, 2025-03-29:
Hello everyone,
I’ve encountered an issue where plot_unit_waveforms from spikeinterface fails to display any waveforms, even though the data is visible in the spikeinterface GUI. Here’s the context:
Workflow
Data: Processed a Plexon file (from 4 tetrodes) using Kilosort4.
Analysis: Successfully computed waveforms and templates via spikeinterface.
Problem: When calling plot_unit_waveforms, the output is blank.
Expected vs. Actual Behavior
Expected: A plot showing spike waveforms for selected units.
Actual: Empty figure with no errors.
```python
extensions_to_compute = [
    "random_spikes",
    "waveforms",
    "principal_components",
    "templates",
    "noise_levels",
    "spike_amplitudes",
    "unit_locations",
    "spike_locations",
    "template_similarity",
    'quality_metrics',
    "template_metrics",
    "isi_histograms",
    "correlograms",
]

extension_params = {
    "unit_locations": {"method": "center_of_mass"},
    "spike_locations": {"ms_before": 0.5,"ms_after":0.5},
    "correlograms": {"bin_ms": 0.1},
    "template_similarity": {"method": "cosine_similarity"},
    "template_metrics":{"include_multi_channel_metrics":True}
}
compute_folder = "compute_KS4_group"
analyzers.compute(
    extensions_to_compute,
    extension_params=extension_params,
    verbose=True
)
saved_analyzer = analyzers.save_as(
    folder=compute_folder,
    format="binary_folder",
)
```
```
compute_waveforms 
engine=process - n_jobs=40 - samples_per_chunk=30,000 - chunk_memory=937.50 KiB - total_memory=36.62 MiB - chunk_duration=1.00s

compute_waveforms (workers: 40 processes): 100%
 17831/17831 [00:06<00:00, 3469.33it/s]
Fitting PCA: 100%
 48/48 [01:35<00:00,  1.99s/it]
Projecting waveforms: 100%
 48/48 [00:00<00:00, 725.05it/s]
noise_level (workers: 20 processes): 100%
 20/20 [00:00<00:00,  2.01it/s]

Compute : spike_amplitudes + spike_locations 
engine=process - n_jobs=40 - samples_per_chunk=30,000 - chunk_memory=937.50 KiB - total_memory=36.62 MiB - chunk_duration=1.00s

Compute : spike_amplitudes + spike_locations (workers: 40 processes): 100%
 17831/17831 [00:09<00:00, 2415.66it/s]

Computing num_spikes
Computing firing_rate
Computing presence_ratio
Computing snr
Computing isi_violation
Computing rp_violation
Computing sliding_rp_violation
Computing amplitude_cutoff
Computing amplitude_median
Computing amplitude_cv
Computing synchrony
Computing firing_range
Computing drift
Computing sd_ratio

noise_level (workers: 20 processes): 100%
 20/20 [00:00<00:00,  2.09it/s]
calculate_pc_metrics: 100%
 48/48 [00:03<00:00, 15.72it/s]

Skipping property 'KSLabel: difference in dtype between sortings'
Skipping property 'KSLabel_repeat: difference in dtype between sortings'
```

[image]
```python
si.plot_unit_waveforms(analyzers,unit_ids=[30])
```

[image]

[image]

zm711, 2025-03-29:
It seems to me like your aptitude is pretty small for this particular unit. Do you have any units with larger amplitudes? I'm wondering if your autoscaling for the `plot_unit_waveforms` is too just flattening your data. So could you try to plot all your different units. If none of them appear then we need to dig more, but if this one doesn't appear then I would lean more toward it being a a scaling issue for a very small amplitude unit.

chrishalcrow, 2025-03-30:
Hello, I have an alternative theory - that we are bad at handling the diamond probe arrangement. 

There's a proposed solution in PR #3748. @cuowunimei , would you be able to try out this new PR? To do so, you need to install spikeinterface from source (https://spikeinterface.readthedocs.io/en/stable/get_started/installation.html), install the github cli and run `gh pr checkout 3748` while in your local spikeinterface repository, then rerun the plotting part of your code?

zm711, 2025-03-30:
I thought we had already merged that PR so I thought this was an issue on top. But agree with Chris. Try the new scaling PR and see if that helps with tetrodes!

chrishalcrow, 2025-04-01:
Hi @cuowunimei , we just merged PR #3748, which should solve this issue. If you install spikeinterface from source (https://spikeinterface.readthedocs.io/en/stable/get_started/installation.html), this installation will contain the bug fix and your plots should work :)