SpikeInterface/spikeinterface #3821: Hugging Face Connectivity Issues

zm711, 2025-03-30:
I wasn't part of the hugging face addition so I'm not sure how important those tests are. we already have to deal with gin and ibl connectivity. Should we do a skip if hugging face is down or do think we need to absolutely check those tests for each PR? Since I didn't follow that specific addition I figured it was better to ask to get clarity for myself :)

alejoe91, 2025-03-30:
We can definitely skip, but I'm not sure how to test if HF is down :)

zm711, 2025-03-30:
Can we ping from the runners? Maybe I'll play around a bit. There's no rush for this :)

samuelgarcia, 2025-04-01:
I would skip and have regular cron ci depending on external platform.

samuelgarcia, 2025-04-02:
About the sortingcomponents, we plan an enormous clean with Pierre in spring.

zm711, 2025-07-14:
The spring cleanup was spring of 2030. I see now. :P