SpikeInterface/spikeinterface #3853: plot_unit_locations error with scaling

jamcdan, 2025-04-10:
I seem to run into a problem when trying to plot unit locations from a Neuropixel UHD probe. The scaling seems to be very off for the electrode positions which squishes the actual data to about 1/6th the size of the probe. I say 1/6th because it might be related to the 6um spacing of the probes and some bug related to that? any help sorting this out is appreciated!
[image]

chrishalcrow, 2025-04-10:
Hi @jamcdan , I suspect the plot is using the "full" probe with all of the electrodes from the NPU device, and you've recorded from the electrodes at the very bottom. Does that sound plausible? Depending on the situation, you might want to plot this, or not. So I think the solution here is to give the user more flexibility in the x/y-limits of the plot (as we do for raster plots and some others). I'll make a PR.

jmcdaniel2, 2025-04-10:
Yeah that sounds likely, I used the 2x192 configuration just at the bottom of the probe

chrishalcrow, 2025-04-10:
Great, PR underway. In the meantime @alejoe91 also points out the following:

"we can also control the limits later:
```
w = sw.plot_unit_locations(...)
w.ax.set_xlim(...)
w.ax.set_ylim(...)
```
"
https://github.com/SpikeInterface/spikeinterface/pull/3855#issuecomment-2792166362

zm711, 2025-04-14:
@jmcdaniel2 did you want to test out the code on main to see if this fixes your issue? Or use the matplotlib strategy? Just trying to figure out if we are good to close :)

jmcdaniel2, 2025-04-14:
I used the matplotlib way and that worked for me. I can check the new code to see if that method works as well

zm711, 2025-04-16:
I close this as the issue itself has been solved. But if you are able to test the PR feel free to comment in the closed issue so future users can see :)

closed 2025-04-16 by zm711