SpikeInterface/spikeinterface #3858: Referencing of read_openephys

jazlynntan, 2025-04-10:
Hello,

Just wanted to clarify the following regarding referencing of Neuropixels 2.0 data as I have an artifact which I suspect is due to a referencing issue.

1. Is the recording object created by read_openephys already referenced based on the selected option in openephys during recording? This would also be what was saved in the settings.xml file as 'referenceChannel' and 'Reference Channel Index'?

2. Does common_reference re-reference the raw unreferenced data to the new selected option (Eg. 'global', 'median') or is it applied onto the referenced recording (if the object created by read_openephys is already referenced)?

3. Is is possible to see the trace of the reference channel (whether it be the external or tip reference)?

Thank you!

alejoe91, 2025-04-11:
Hi @jazlynntan 

1. The `read_openephys` reads the `continuous.dat` file. Since the referencing of NP probes is on chip, I think that the traces saved by Open Ephys are already referenced.

2. The CMR is applied to the continuous dat, so "it doesn't know" about any previous referencing

3. Only if you used an internal channel as reference. As far as I know, the tip reference or an external reference are not recorded.

@jsiegle @jenniferColonell can confirm

jazlynntan, 2025-04-11:
I see, thank you!

jenniferColonell, 2025-04-11:
Hi Jazlynn,
Just want to confirm Alessio's answers, with one more detail with respect to point 3: On the commercial 2.0 probes, there are no internal site references (because they don't work very well).

Also, I want to repeat that tip referencing is generally not recommended for LFP measurements because the result for all sites on the probe is the difference between the local LFP and the LFP near the tip. The result is an apparent lack of any LFP for sites near the tip, and distorted LFP elsewhere. Tip referencing only works well for spiking data because the associated signals have very small spatial extent -- LFP signals extend over 100's of microns.