SpikeInterface/spikeinterface #3861: Error running sorting with kilosort v4.0.31

JarrodDowdall, 2025-04-14:
I noticed in the latest version of Kilosort v4.0.31

`cluster_spikes()` now returns 4 outputs:

https://github.com/MouseLand/Kilosort/blob/7ddd8a2dc326fa770960a4494546d44b3548a4ed/kilosort/run_kilosort.py#L830

```
def cluster_spikes(st, tF, ops, device, bfile, tic0=np.nan, progress_bar=None,
                   clear_cache=False, verbose=False):
....
return clu, Wall, st, tF
```

But `_run_from_folder()` in `Kilosort4Sorter()` still expects two outputs:

https://github.com/SpikeInterface/spikeinterface/blob/a376f50089ddf6ffd23eef2b44bb211dcd58919a/src/spikeinterface/sorters/external/kilosort4.py#L361

which leads to the error `ValueError: too many values to unpack (expected 2)`

alejoe91, 2025-04-14:
Hi @JarrodDowdall 
We have our internal tests failing as well, and they usually tell us that something has changed on the KS4 side :) Give us a week or so to understand what's different and push some changes! For the time being, please stock to 4.0.30, as I don't think they new version is going to have a major impact on performance

darioringach, 2025-04-18:
Yup -- same problem here.

closed 2025-04-18 by alejoe91