SpikeInterface/spikeinterface #3865: Question `commen_reference` for multi-shank probes:

chiyu1203, 2025-04-15:
I had a brief conversation with the CambridgeNeurotech about `commen_reference` today. It seems that since I am using a multi-shank probe, it would be great to take channels from one shank as a reference for channels from the other shank, and the other way round (instead of calculating the reference based on all channels). Has anyone tried this method before? In addition, I understand that current version of `commen_reference` can take dictionary as an input, which helps process data with multi-shank probes. In this case, how does the function work when those reference methods: `"global" | "single" | "local"` are called? I was wondering if the easiest way to take this concern into account when taking the reference would be giving "local_radius" abd not splitting channels into dictionary, so that for every channels there would be an unique combination of channels set for calculating common reference (depending on the radius). Does it make sense?

h-mayorquin, 2025-04-16:
Relevant:

https://spikeinterface.readthedocs.io/en/stable/how_to/process_by_channel_group.html

chiyu1203, 2025-04-16:
Thanks for pointing that article out! That's what I meant when I wrote "current version of commen_reference can take dictionary as an input", In my analysis pipeline, I first split channels by shank and stored them as a dictionary so that they can be processed by those function  `commen_reference` and `whiten` shank by shank already. However, if I keep using the default setting for `commen_reference`, I will basically use all channels from the same shank for calculating its own reference, which is the opposite from what CambridgeNeurotech expert suggested. That's why I thought setting local_radius is the easiest way to go but wondered if anyone has experience with this already.

zm711, 2025-04-19:
I think this is an interesting idea. But I'm wondering how we would implement this in the case of more than two shanks? I would love for @alejoe91 or @samuelgarcia  to comment. I think this would also be influenced by how far away the shanks are from each other. If you're in two different tissue types with the shanks then I don't know if cross-shank would actually be a good reference.