SpikeInterface/spikeinterface #3869: Concatenating Open Ephys recordings with time gaps: insert zero padding to maintain continuity before concatenate

castro-lab, 2025-04-16:
Hello,
In OE if the recording was paused there will be time gaps between segments... Is there a way to fill segment time gaps before concatenate_recordings (which eliminates time) so that, for example, a full continuous OE "experiment" with filled gaps between "recordings" can be sent to kilosort4 to preserve time continuity?  Any simple way to fill these segment gaps with zero padding in si before concatenate_recprdings?

rec = si.read_openephys(folder_path=exp_folder)
rec = si.concatenate_recordings([rec])

zm711, 2025-04-16:
I vaguely remember this discussion happening for one of the readers (but I can't remember which). but it ended up being more of a mess than we realized. This is because we have to decide if this is a true 0 vs an unscaled 0 (which might end up scaling to a non-zero value) and you could create strange edges where you end up with spikes at those discontinuities junctions. We often see that KS versions with their padding end up generating spikes before and after recordings (which I hypothesize is due to the padding) so this is really a non-trivial problem. Maybe @samuelgarcia and @alejoe91 remember this better. But I think I would be against this now that I've seen how many weird things it can introduce, but maybe others are more familiar and think it is worth this.

castro-lab, 2025-04-16:
thanks. How is SI padded when you add artifact intervals to be excluded periods from sorting? I guess this could use the same padding as in that case. I scripted a function that seems to fill the gaps correctly, creating a final 1-segment Rec for KS, but I am using 0 as the filler. 

These gap periods are irrelevant and only used for time continuity, meaning that any spikes found there will be excluded from subsequent analyses (so this should not be a problem as long as this does not affect the KS sorting of the other periods).

zm711, 2025-04-16:
If you don't care about these times why don't you just concatenate? You have stimulus data that you are worried about for the timing? Your script above would be my rec and not worry about the gaps. If you already know the time to pad though couldn't you just correct stim times such that if KS returns sample 1000 you know that due to the gap you just change it to sample 1050?

castro-lab, 2025-04-16:
This is not a sole OE recording, but there are several other things that were also recorded and follow the full experiment timeline. There are multiple gaps and several associated files. The simplest way is to keep the original timeline in the OE experiment. If you concatenate, you lose that timeline. The best approach in this case is to fill the gaps and keep the timeline. 

I can get a segment from the rec
        **segment = rec.get_traces(segment_index=i,return_scaled=False)**
I can create a filler
            **filler_data = np.zeros((gap_frames, num_channels), dtype=dtype)
            filler = si.NumpyRecording(traces_list=filler_data, sampling_frequency=fs, channel_ids=channel_ids)**

and then concatenate as needed in order (multiple times). This works fine and creates a correctly sized rec, but then I get the error 1 when applying **si.detect_bad_channels(rec)**. If I set **return_scaled=true**, then I get an error 2 when I try to concatenate the filler with the segment because they do not have the same attributes (ch ids, etc). Is there a way to create the **filler numpy** (with zeroes) so that it matches all the existing attributes of the rec segments (channels, gains, offsets) so it can be concatenated with the segment, keeping the segment attributes?  Also, it does not seem that **si.detect_bad_channels(rec)** can be set to **return_scale=False** as it does not have that variable. So this must be an internal error? 

ERROR1:
    bad_channel_ids, channel_labels= si.detect_bad_channels(rec)
                                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: This recording does not support return_scaled=True (need gain_to_uV and offset_to_uV properties)

ERROR2:

ValueError: Recording don't have the same dtype or channel_ids

zm711, 2025-04-18:
@castro-lab I'm busy for the next couple days. could you ping me on Monday so I can read this carefully. Unless @alejoe91 has a moment to take a look at this one :)

alejoe91, 2025-04-18:
@castro-lab there is a better way to do what you are looking for :)

I would concatenate all recordings as you do, run kilosort, and than split back the sorting objects in segments and handle time separately for them. This will minimally affect spike sorting (there might be some minor filtering effects at the edges) and overall make your life easier. Actually, filling in zeros for long periods might mess up with estimates of noises throughout the recordings...

Here's an example:
```
# load multi-segment recording
rec = si.read_openephys(folder_path=exp_folder, load_sync_timestamps=True)
# the load sync timestamps will read the open ephys timestamps and assign it to the recording!

# concatenate
rec_concat = si.concatenate_recordings([rec])

# run kilosort
sort_concat = ss.run_sorter(rec_concat, "kilosort4", ...)

# split back the sorting back to multi-segment
sort = si.split_sorting(sort_concat, rec_concat)

# register original recording to split sorting
sort.register_recording(rec)
```

Now that both your recording and sorting are multi-segment, you can get the spike times (in seconds) of your spikes with:
```
spike_times = sort.get_unit_spike_train(unit_id=sort.unit_ids[0], return_times=True)
```
The `return_times=True` will give you the timestamp from the recording associated to the spike samples.

castro-lab, 2025-04-19:
In some cases, I have multiple OE "experiments" (same ch locations) with each containing a few "recordings" (segments), and would like to sort them together (combining multiple experiments). 

Here, I append the experiments (using a list of already read objects), then concatenate for sorting, split, and register using the **rec_append**:
```
        rec_append = si.append_recordings(group_recordings)  # append list of Experiments
        rec_concat = si.concatenate_recordings(group_recordings)  # Combine for spike sorting     
        sort_concat = run_spike_sorting(rec_concat, group_sorter_output, sorter_name, gpu1)       
        sorting = si.split_sorting(sort_concat, rec_append)  # Split by original segments
        sorting.register_recording(rec_append)  # Associate sorting with the experiments 
```
If correct, how do I now get the units of my 1st or 2nd experiment? I can specify segments below, but in this case, I really want to specify experiments (per OE). 
`    spike_vector = sorting.to_spike_vector()`

alejoe91, 2025-04-21:
When appending recordings, you are going to have a single recording with N_experiment * N_recording segments. So if you have, say, two experiments with two recording each, segments 0-1 will be from exp 1, and segmtent 2-3 from exp 2. Note that since you're concatenating, you will have the same units across experiments!

castro-lab, 2025-04-21:
Yes, I understand the difference between append and concatenate in SI (well explained on the API). I am no longer using append but do concatenate directly:

First, get a list of the experiments into group_recordings and use it here:
```
rec_concat = si.concatenate_recordings(group_recordings)    # Combine for spike sorting 
sort_concat = run_spike_sorting(rec_concat, group_sorter_output, sorter_name, gpu1)
sorting = si.split_sorting(sort_concat, rec_concat)   # split by original segments

```
Then loop on the original experiments in group_recordings, getting each rec from the previous list, and do:
```
            num_segments = rec.get_num_segments()                 # num of segments in experiment
            segment_indices = list(range(segment_idx, segment_idx + num_segments)) # get segment indices
            experiment_sorting = si.select_segment_sorting(sorting, segment_indices)
            experiment_sorting.register_recording(rec)
            segment_idx += num_segments
```
            in each loop, output spikes for each experiment

This does the job by rebuilding the segments with si.select_segment_sorting and registering with each original experiment=rec... What do you think?

alejoe91, 2025-04-22:
Yep that should work!

alejoe91, 2025-04-22:
Awesome! Closing the issue then :)

closed 2025-04-22 by alejoe91