SpikeInterface/spikeinterface #3881: Question regarding read_intan()

OlivierPeron, 2025-04-23:
Hi, 

I was using the function `read_intan()` to read an rhd file, with neo version= 0.13, but I have now updated the package and the function doesn't work. It seems to look for several intan files, such as the time.dat file, which shouldn't be necessary to read the info.rhd file. Also all stream_index are being loaded, even_tho a `stream_id `or `stream_name` is given. 

I am running :
```
intan_info = se.read_intan(paths['rhd'], stream_id = '0')
```

and I got the following error, due to trying to check all streams :

```
intan_info = se.read_intan(paths['rhd'], stream_id = '0')
  File "C:\Users\DeepLabCut\Anaconda3\envs\spykeline\lib\site-packages\spikeinterface\extractors\neoextractors\intan.py", line 58, in __init__
    NeoBaseRecordingExtractor.__init__(
  File "C:\Users\DeepLabCut\Anaconda3\envs\spykeline\lib\site-packages\spikeinterface\extractors\neoextractors\neobaseextractor.py", line 191, in __init__
    _NeoBaseExtractor.__init__(self, block_index, **neo_kwargs)
  File "C:\Users\DeepLabCut\Anaconda3\envs\spykeline\lib\site-packages\spikeinterface\extractors\neoextractors\neobaseextractor.py", line 27, in __init__
    self.neo_reader = self.get_neo_io_reader(self.NeoRawIOClass, **neo_kwargs)
  File "C:\Users\DeepLabCut\Anaconda3\envs\spykeline\lib\site-packages\spikeinterface\extractors\neoextractors\neobaseextractor.py", line 69, in get_neo_io_reader
    neo_reader.parse_header()
  File "C:\Users\DeepLabCut\Anaconda3\envs\spykeline\lib\site-packages\neo\rawio\baserawio.py", line 211, in parse_header
    self._parse_header()
  File "C:\Users\DeepLabCut\Anaconda3\envs\spykeline\lib\site-packages\neo\rawio\intanrawio.py", line 297, in _parse_header
    tmp.append(raw_data[0].size)
IndexError: list index out of range
``` 

with :
``` 
{'RHD2000 amplifier channel': [], 'RHD2000 auxiliary input channel': [], 'USB board digital input channel': [], 'timestamp': [memmap([       0,        1,        2, ..., 12559741, 12559742, 12559743])]}
```

h-mayorquin, 2025-04-23:
Hi,

This is a neo issue. Can you open something there? Please state your version.

Also, is that your full traceback? I am unable to locate the instruction that is making your code fail on the code with what you shared.

OlivierPeron, 2025-04-24:
Hi,

Ok will do ! My neo version is 0.14.1 !

Yes, the instruction is `intan_info = se.read_intan(paths['rhd'], stream_id = '0')`.

OlivierPeron, 2025-04-24:
Solved this in issue #1693 of the python-neo repo.

closed 2025-04-24 by OlivierPeron

h-mayorquin, 2025-04-24:
The issue here was the user only had the `amplifier.dat` file and not the output as the acquisition system as it comes. In this case, the one file per signal, the format is packed binary and the `BinaryRecordingExtractor` could be used to load data:

https://spikeinterface.readthedocs.io/en/stable/how_to/load_matlab_data.html#loading-data-in-spikeinterface

It should be pretty straightforward to get the necessary arguments from the sepc:
https://intantech.com/files/Intan_RHS2000_data_file_formats.pdf#page=11.05

And also some more useful comments here about the binary structure that would apply when reading from python:

https://github.com/NeuralEnsemble/python-neo/pull/1558#pullrequestreview-2313518314

OlivierPeron, 2025-04-25:
> The issue here was the user only had the `amplifier.dat` file and not the output from the acquisition system as it comes.
> 
> In this case, the "one file per signal", the format is packed binary and the `BinaryRecordingExtractor` could be used to load data:
> 
> https://spikeinterface.readthedocs.io/en/stable/how_to/load_matlab_data.html#loading-data-in-spikeinterface
> 
> It should be pretty straightforward to get the necessary arguments from the sepc: https://intantech.com/files/Intan_RHS2000_data_file_formats.pdf#page=11.05
> 
> [image]
> 
> And also some more useful comments here about the binary structure that would apply when reading from python:
> 
> [NeuralEnsemble/python-neo#1558 (review)](https://github.com/NeuralEnsemble/python-neo/pull/1558#pullrequestreview-2313518314)

I wanted to open the info.rhd file, not the amplifier.

h-mayorquin, 2025-04-25:
Intan themselves provide routines for reading the headers:

https://intantech.com/downloads.html?tabSelect=Software&yPos=0