SpikeInterface/spikeinterface #3886: Unable to run the docker image for waveclus

yuerout, 2025-04-26:
Hi! I am trying to run the image for [waveclus](https://hub.docker.com/r/spikeinterface/waveclus-compiled-base). However, when I did
```
sorting_waveclus = si.run_sorter(sorter_name='waveclus', recording=raw_rec_sub, remove_existing_folder=True, docker_image=True,
                             output_folder="homedirectory/waveclus_output",
                             verbose=True, installation_mode="pypi")
```
It tells me that

```
---------------------------------------------------------------------------
SpikeSortingError                         Traceback (most recent call last)
Cell In[20], line 3
      1 sorter_params = {'do_correction': False}
      2 # run spike sorting on entire recording
----> 3 sorting_KS25 = si.run_sorter(sorter_name='waveclus', recording=raw_rec_sub, remove_existing_folder=True, docker_image=True,
      4                              output_folder="/homedirectory/neuron/waveclus_output",
      5                              verbose=True, installation_mode="pypi")

File ~/anaconda3/envs/spike-sorting/lib/python3.11/site-packages/spikeinterface/sorters/runsorter.py:193, in run_sorter(sorter_name, recording, folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, delete_container_files, with_output, output_folder, **sorter_params)
    187         if not has_spython():
    188             raise RuntimeError(
    189                 "The python `spython` package must be installed to "
    190                 "run singularity. Install with `pip install spython`"
    191             )
--> 193     return run_sorter_container(
    194         container_image=container_image,
    195         mode=mode,
    196         **common_kwargs,
    197     )
    199 return run_sorter_local(**common_kwargs)

File ~/anaconda3/envs/spike-sorting/lib/python3.11/site-packages/spikeinterface/sorters/runsorter.py:648, in run_sorter_container(sorter_name, recording, mode, container_image, folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, delete_container_files, extra_requirements, installation_mode, spikeinterface_version, spikeinterface_folder_source, output_folder, **sorter_params)
    646 if run_error:
    647     if raise_error:
--> 648         raise SpikeSortingError(f"Spike sorting in {mode} failed with the following error:\n{run_sorter_output}")
    649 else:
    650     if with_output:

SpikeSortingError: Spike sorting in docker failed with the following error:
Traceback (most recent call last):
  File "/homedirectory/neuron/in_container_sorter_script.py", line 4, in <module>
    from spikeinterface import load
ModuleNotFoundError: No module named 'spikeinterface'
```

I started the container with `docker run -i -t spikeinterface/waveclus-compiled-base  /bin/bash` and installed spikeinterface with pip but it give me the same error. 

I moved on to another method, which was to create a custom Docker image. I created it with 
```
FROM spikeinterface/waveclus-compiled-base:latest
RUN pip install "spikeinterface"
RUN pip install "h5py==2.10.0" # i got an error message which tells me to downgrade h5py
```
I then used this image for run_sorter, and it tells me that 
```
---------------------------------------------------------------------------
SpikeSortingError                         Traceback (most recent call last)
Cell In[19], line 3
      1 sorter_params = {'do_correction': False}
      2 # run spike sorting on entire recording
----> 3 sorting_KS25 = si.run_sorter(sorter_name='waveclus', recording=raw_rec_sub, remove_existing_folder=True, docker_image="waveclus-with-si-0110:0.1.0",
      4                              output_folder="/homedirectory/neuron/waveclus_output",
      5                              verbose=True, installation_mode="pypi")

File ~/anaconda3/envs/spike-sorting/lib/python3.11/site-packages/spikeinterface/sorters/runsorter.py:193, in run_sorter(sorter_name, recording, folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, delete_container_files, with_output, output_folder, **sorter_params)
    187         if not has_spython():
    188             raise RuntimeError(
    189                 "The python `spython` package must be installed to "
    190                 "run singularity. Install with `pip install spython`"
    191             )
--> 193     return run_sorter_container(
    194         container_image=container_image,
    195         mode=mode,
    196         **common_kwargs,
    197     )
    199 return run_sorter_local(**common_kwargs)

File ~/anaconda3/envs/spike-sorting/lib/python3.11/site-packages/spikeinterface/sorters/runsorter.py:648, in run_sorter_container(sorter_name, recording, mode, container_image, folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, delete_container_files, extra_requirements, installation_mode, spikeinterface_version, spikeinterface_folder_source, output_folder, **sorter_params)
    646 if run_error:
    647     if raise_error:
--> 648         raise SpikeSortingError(f"Spike sorting in {mode} failed with the following error:\n{run_sorter_output}")
    649 else:
    650     if with_output:

SpikeSortingError: Spike sorting in docker failed with the following error:
Traceback (most recent call last):
  File "/homedirectory/neuron/in_container_sorter_script.py", line 4, in <module>
    from spikeinterface import load
ImportError: cannot import name 'load' from 'spikeinterface' (/home/miniconda3/lib/python3.8/site-packages/spikeinterface/__init__.py)
```

Additionally, while reading #3502 I also realized that I need spikeinterface==0.101.0 but not the latest version, because the waveclus image runs on Python 3.8. I changed the installation to `RUN pip install "spikeinterface[full]==0.101.0"` but it couldn't build this time. 
```
13.47 ERROR: Could not find a version that satisfies the requirement cuda-bindings~=12.8.0 (from cuda-python; platform_system != "Darwin" and extra == "full"->spikeinterface[full]==0.101.0) (from versions: none)
13.63 ERROR: No matching distribution found for cuda-bindings~=12.8.0 (from cuda-python; platform_system != "Darwin" and extra == "full"->spikeinterface[full]==0.101.0)
```

I'd appreciate it if anyone have any insights on how to solve it, or can share any recent successes with using waveclus! Thank you

zm711, 2025-04-26:
My summary reading of this makes me think that this will be a bit tricky since python 3.8 is EOL so you would need to not only limit spikeinterface, but other packages during the python installation in the docker container. It is definitely possible, but it might require some try-error to figure out. I always leave the definitive answer to @alejoe91 as he knows the spikeinterface docker stuff.

yuerout, 2025-05-01:
@zm711 Thanks! Yes, I just made the waveclus sorter run. It might not be the best way but it's working for now. This is what I did:
1. Downgraded the spikeinterface package in my conda environment to `0.101.0`.
2. My dockerfile:
```
FROM spikeinterface/waveclus-compiled-base:latest

RUN pip install "cuda-python==12.2.0"
RUN pip install "spikeinterface[full]==0.101.0"
RUN pip install "h5py==2.10.0"
RUN pip install "scipy==1.10.1"
```
3. run sorter with 
```
si.run_sorter(sorter_name='waveclus', recording=raw_rec_sub, remove_existing_folder=True, docker_image="name-of-docker-image",
                             output_folder="waveclus_output",
                             verbose=True, installation_mode="pypi")
```

zm711, 2025-05-01:
Looks pretty good to me. I'll close this for now although we should probably this at a maintenance meeting--(tweaking our docker building has been a todo for a while). But if you run into any other problems definitely open new issues :)

closed 2025-05-01 by zm711