SpikeInterface/spikeinterface #3894: Incorrect assumptions

billkarsh, 2025-04-29:
Various comments in neuropixels_utils.py regarding the number of ADCs on probes are wrong. Have not checked if code is also wrong.

zm711, 2025-04-29:
Thanks for taking the time to look at the project, we always appreciate when people involved in the various tooling provide feedback. That being said it would be more helpful if you pointed out which assumptions are wrong so we can fix them. Here's a [nifty article](https://dev.to/opensauced/how-to-write-a-good-issue-tips-for-effective-communication-in-open-source-5443)  on how to write a good issue. And we always welcome PRs :)

billkarsh, 2025-04-29:
That's cute. Given what I told you that should trigger a full code review. It's best if you look at everything. At least review everything regarding time shifting and test it.

billkarsh, 2025-04-30:
The operations are documented here: https://github.com/billkarsh/CatGT/blob/main/Build/Tool.cpp::FFT::init(), FFT::timeShiftChannel().

The mux tables are in the public probeTable repo: https://github.com/billkarsh/ProbeTable.

The description of the mux table is here: https://billkarsh.github.io/SpikeGLX/Sgl_help/Metadata_30.html#imec.

The mux table is organized as (nADC,nGrp)(grp)(grp)… Each (grp) are the (acquired) channel indices that are digitized together.

The shift amount delT(igrp) is the time difference between group-igrp and group-0. Group-0 does not need to be shifted.

delT(igrp) = igrp/(nGrp*sample_rate).

Note that for 2.0-like probes (that are full band only), nGrp should be used as is from the table. In these cases the multiplexer clock runs at rate nGrp*sample_rate. That is, nGrp sets of channels are digitized during each effective sample period (1/sample_rate).

However, for 1.0-like probes that have an AP and an LF band, the multiplexer runs at (nGrp+1)*ap_sample_rate. In all cases so far, nGrp=12. During each sample period, the multiplexer digitizes {nGrp=12 AP channel groups + 1 LF group}. The LF groups are the same groups as the AP groups in the table, however, it takes 12 AP sample periods to digitize all of the LF groups. That’s how the LF sample rate becomes exactly 1/12 of the AP sample rate.

PR #3895 (2025-05-07): Clarify docstrings in neuropixels_utils
Fixes #3894

@billkarsh Do you mind taking a look? I believe that the main error and confusion point was in the use of `num_channels_per_adc` and its description
closed 2025-05-07 by samuelgarcia