SpikeInterface/spikeinterface #3901: SpikeInterface and Kilosort4 compatibility

JulienBrn, 2025-05-05:
Hi,

Kilosort4 seems to change very quickly and I seem unable to find a combination of spikeinterface/kilosort4 versions with which I do not encounter problems (but the problems change depending on the version combination). It would be really nice to have within the docs the version of kilosort with which the version of spikeinterface has been tested with.

As for my current compatibility issue, I was trying with spikeinterface 0.102.2 and kilosort 4.0.32 (April 16th, but there's already 4 new pypi releases since then...). I have 2 problems:

1. I had to change the code of spikeinterface because the API of cluster_spikes has changed
2. The code attempts to allocate 150Gb of memory

The relevant code is below and the spikeinterface log is attached. As I see it, my options are the following:

1. Use an older version of kilosort (that should at least solve probem 1)) and just hope that it also solves problem 2). But which version to use (I had already attempted another version pair previously with other problems) ?
2. Just wait for a new version of spikeinterface and hope for the best
3. Attempt to solve problem 2) by modifying the code myself, but this is not maintainable long term.

```python
#MY MODIFICATION
clusters = cluster_spikes(**cluster_spikes_kwargs)
try:
    clu, Wall = clusters
except Exception:
    clu, Wall, _, _ = clusters #Kilosort now returns a 4 element tuple...
#End of my modification

if params["skip_kilosort_preprocessing"]:
    ops["preprocessing"] = dict(
        hp_filter=torch.as_tensor(np.zeros(1)), whiten_mat=torch.as_tensor(np.eye(recording.get_num_channels()))
    )

print("Before save sorting", st.shape, clu.shape, Wall.shape) #Just to see

#This fails (trying to allocate 150Gb) as we have 15*10^6 spikes and near 400 channels

_ = save_sorting(
    ops=ops,
    results_dir=results_dir,
    st=st,
    clu=clu,
    tF=tF,
    Wall=Wall,
    imin=bfile.imin,
    tic0=tic0,
    save_extra_vars=save_extra_vars,
    save_preprocessed_copy=save_preprocessed_copy,
)

####This happends in kilosort save to phy line 420 of io.py

pc_features, pc_feature_ind = make_pc_features(
        ops, spike_templates, spike_clusters, tF
        )

#### Which calls (indirectly) get_data_cpu of clustering_qr.py

dd = torch.zeros((nspikes, nchan, nfeatures)) #Which is too big for memory
```

[spikeinterface_log.json](https://github.com/user-attachments/files/20036556/spikeinterface_log.json)

zm711, 2025-05-05:
@JulienBrn,

you've hit the nail on the head. KS4 is changing so fast that it's hard for us to keep up. We are nearly all the way caught up on main but if you are installing spikeinterface from pypi then it will be hard to sync things up . (See PR #3900 which is close to being merged and should bring us up to the most recent KS4--unless it changes again quickly). 

As far as indicating which KS4 works with which spikeinterface, I think that is a hard problem to indicate. Some changes are internal to KS4 so our wrapper doesn't need to change the user just needs to update and other changes require fixes at our wrapper layer. We can't predict which is which so we just have to wait for users to report that KS4 has changed and then we fix it.

That being said for your specific issues:
1) do you install from github or from pypi? If you are worried about changes then installing spikeinterface from main will allow you to get our updated wrappers much faster
2) the `save_sorting` function of kilosort is not under our control. If it is taking up too much GPU or CPU memory that would be a bug report for KS4 so they can work on optimizing it--but if you are running memory profiling and have tracked it down to our code let us know which function is causing the issue so we can try to troubleshoot.

Does that all make sense?

JulienBrn, 2025-05-05:
Thank you for the in depth answer. All this makes sense but:

1. Although keeping track of all compatibility may be hard, just specifying in the docs (when pushed on main) a (not all) version (pypi or git) for which it has been tested as compatible would be a nice addition : one could start there, check that it works and then attempt to update kilosort if one feels the need, which may then work or fail.
2. I've installed from main. Though for the moment I'm not so much interested in getting the most up to date version, but more in just getting a stable pipeline that works (we are migrating from spikeinterface + kilosort2.5 to kilosort4). Once I have a pipeline with kilosort4 working, I'll consider how often we would want to update (perhaps every 6 months?).
3. As for the save sorting function, it is indeed mostly handled by kilosort. However, my understanding is that the save function of kilosort saves much more information then what is required by spikeinterface (waveforms, ...) and that the sorting analyzer then recomputes that information anyway. Perhaps there is no need to call the kilosort save sorting function as spike interface is already using cluster_spikes directly?

(and thank you for the link showing the tests)

zm711, 2025-05-05:
> Although keeping track of all compatibility may be hard, just specifying in the docs (when pushed on main) a (not all) version (pypi or git) for which it has been tested as compatible would be a [...]

@alejoe91 what do you think about adding this somewhere in a docstring? We could say validated up to version >xx please report an issue if a problem occurs with a later version? 

> I've installed from main. Though for the moment I'm not so much interested in getting the most up to date version, but more in just getting a stable pipeline that works (we are migrating from [...]

Fair enough. I think it's great to get the migration going! As someone who follows the development I update way more often, but usually there are a few releases a year so maybe even quarterly updates once you have your pipeline stabilized. We are constantly trying to optimize (+ the normal bug fix stuff), so I'm pro-being uptodate.

> As for the save sorting function, it is indeed mostly handled by kilosort. However, my understanding is that the save function of kilosort saves much more information then what is required by [...]

This is another great question. We do this because we have some users that only want to use spikeinterface for preprocessing and sorting, we want to return the "true" sorting results such that someone could act as though they just ran the sorter without our wrapper. That being said we think our postprocessing and qualitymetrics are super cool and useful (so we hope everyone wants to use them). Let's ping @alejoe91 and see if he has any interest in allowing us to have a kwarg that would do a spikeinterface save vs a full KS4 save. It would add maintenance burden, but I don't personally see a problem with it.

danielsvedberg, 2025-05-13:
Just wanted to throw out there that I just managed to successfully sort with KS4 via spikeinterface so I'm going to share my versioning/install in the hopes this is at least somewhat helpful to anyone else trying to get SI and KS4 to play nice.

Machine/OS: Win11, running KS4 using NVIDIA GPU.  

The roundabout way I installed, for posterity/reproducibility:

- initially installed spikeinterface into a conda environment via the[ beginner's installation guide ](https://github.com/SpikeInterface/spikeinterface/tree/main/installation_tips)
- After facing some sorting failures running KS4 in docker, I decided to reinstall spikeinterface 0.102.4 installed via git clone. 
- After continued failures, I went for the hail-mary option of installing KS4 4.0.36 in the spikeinterface conda environment via `pip install kilosort`, followed by reinstallation of pytorch 2.7.0 via `pip install torch --index-url https://download.pytorch.org/whl/cu118` (per ks4's install instructions)
- by some miracle, this worked without throwing errors

Here's my .yml file text pasted into the comment since github won't let me attach .yml file (very sorry for the wall of text):

```
name: si_env
channels:
  - conda-forge
  - https://repo.anaconda.com/pkgs/main
  - https://repo.anaconda.com/pkgs/r
  - https://repo.anaconda.com/pkgs/msys2
dependencies:
  - _openmp_mutex=4.5=2_gnu
  - anyio=4.9.0=pyh29332c3_0
  - argon2-cffi=23.1.0=pyhd8ed1ab_1
  - argon2-cffi-bindings=21.2.0=py311he736701_5
  - arrow=1.3.0=pyhd8ed1ab_1
  - asttokens=3.0.0=pyhd8ed1ab_1
  - async-lru=2.0.5=pyh29332c3_0
  - attrs=25.3.0=pyh71513ae_0
  - babel=2.17.0=pyhd8ed1ab_0
  - beautifulsoup4=4.13.4=pyha770c72_0
  - bleach=6.2.0=pyh29332c3_4
  - bleach-with-css=6.2.0=h82add2a_4
  - brotli=1.1.0=h2466b09_2
  - brotli-bin=1.1.0=h2466b09_2
  - brotli-python=1.1.0=py311hda3d55a_2
  - bzip2=1.0.8=h2466b09_7
  - ca-certificates=2025.4.26=h4c7d964_0
  - cached-property=1.5.2=hd8ed1ab_1
  - cached_property=1.5.2=pyha770c72_1
  - cairo=1.18.4=h5782bbf_0
  - certifi=2025.1.31=pyhd8ed1ab_0
  - cffi=1.17.1=py311he736701_0
  - charset-normalizer=3.4.2=pyhd8ed1ab_0
  - cloudpickle=3.1.1=pyhd8ed1ab_0
  - colorama=0.4.6=pyhd8ed1ab_1
  - comm=0.2.2=pyhd8ed1ab_1
  - contourpy=1.3.2=py311h3257749_0
  - cpython=3.11.12=py311hd8ed1ab_0
  - crc32c=2.7.1=py311he736701_1
  - cycler=0.12.1=pyhd8ed1ab_1
  - debugpy=1.8.14=py311hda3d55a_0
  - decorator=5.2.1=pyhd8ed1ab_0
  - defusedxml=0.7.1=pyhd8ed1ab_0
  - deprecated=1.2.18=pyhd8ed1ab_0
  - donfig=0.8.1.post1=pyhd8ed1ab_1
  - double-conversion=3.3.1=he0c23c2_0
  - exceptiongroup=1.2.2=pyhd8ed1ab_1
  - executing=2.2.0=pyhd8ed1ab_0
  - font-ttf-dejavu-sans-mono=2.37=hab24e00_0
  - font-ttf-inconsolata=3.000=h77eed37_0
  - font-ttf-source-code-pro=2.038=h77eed37_0
  - font-ttf-ubuntu=0.83=h77eed37_3
  - fontconfig=2.15.0=h765892d_1
  - fonts-conda-ecosystem=1=0
  - fonts-conda-forge=1=0
  - fonttools=4.57.0=py311h5082efb_0
  - fqdn=1.5.1=pyhd8ed1ab_1
  - freetype=2.13.3=h57928b3_1
  - glib=2.84.1=h3d4babf_0
[... 223 lines cut ...]
      - mearec==1.9.2
      - meautility==1.5.2
      - mpmath==1.3.0
      - narwhals==1.38.0
      - neo==0.14.1
      - numpy==1.26.4
      - openpyxl==3.1.5
      - probeinterface==0.2.28
      - pynwb==3.0.0
      - quantities==0.16.2
      - ruamel-yaml==0.18.10
      - ruamel-yaml-clib==0.2.12
      - seaborn==0.13.2
      - simplejson==3.20.1
      - skops==0.11.0
      - sortingview==0.14.1
      - spikeinterface==0.102.4
      - spikeinterface-gui==0.10.0
      - spython==0.3.14
      - sympy==1.14.0
      - tabulate==0.9.0
      - torch==2.7.0+cu118
      - tridesclous==1.6.8
      - zarr==2.18.7 
```

JulienBrn, 2025-05-16:
I fixed my memory error by not using the save function from kilosort. I did this by modifying the spike interface code for run kilosort. 

Mainly, I added the following instead of calling the save sorting from kilosort in the _run_from_folder function:
```python
np.save(results_dir / "st.npy", st)
np.save(results_dir / "clu.npy", clu)
import json
with (results_dir / "info.json").open("w") as f:
    json.dump(dict(fs=recording.sampling_frequency), f)
```

And I changed the _get_result_from_folder function to:
```python
import json
sorter_output_folder = Path(sorter_output_folder)
if (sorter_output_folder.parent / "spikeinterface_params.json").is_file():
    params_file = sorter_output_folder.parent / "spikeinterface_params.json"
else:
    # back-compatibility
    params_file = sorter_output_folder / "spikeinterface_params.json"
with params_file.open("r") as f:
    sorter_params = json.load(f)["sorter_params"]
# keep_good_only = sorter_params.get("keep_good_only", False)
import numpy as np
st = np.load(sorter_output_folder/"st.npy")
clu = np.load(sorter_output_folder/"clu.npy")
with (sorter_output_folder/"info.json").open("r") as f:
    info = json.load(f)
sorting = NumpySorting.from_samples_and_labels(st[:, 0], clu, info["fs"])
#sorting = KiloSortSortingExtractor(folder_path=sorter_output_folder, keep_good_only=keep_good_only)
return sorting
```

I believe these modifications to be far from optimal (iḿ not handling the "keep good only" parameter for example), but they have the advantage of working. I'm hoping a parameter about how to save can be added in the "run_sorter" function from spikeinterface so that we do not need to rely on the sorter's save function when not necessary.

samuelgarcia, 2025-07-02:
+1 for having a option to do `save_sorting` or not