SpikeInterface/spikeinterface #3902: temp_wh.dat and recording.dat

anhknguyen96, 2025-05-06:
Hi, I've just recently used the package so it's still quite a learning curve for me. Is it normal to not have temp_wh.dat and/or recording.dat in the sorter output (kilosort4 run as wrapper)?
Thank you for the awesome work btw!

zm711, 2025-05-06:
Also just for your future use we have these two functions to help people understand sorter params.
https://github.com/SpikeInterface/spikeinterface/blob/c4bf17e69f0660b7145511640f8b8608925cf604/src/spikeinterface/sorters/sorterlist.py#L79-L124. 

so you could do:

```python

from spikeinterface.sorters import get_default_sorter_params, get_sorter_params_description

get_default_sorter_params("kilosort4")
get_sorter_params_description("kilosort4")
```

and that will return the info I linked above + the descriptions of what each of those params mean.

anhknguyen96, 2025-05-06:
thank you for the very prompt and thorough replies!!

anhknguyen96, 2025-05-10:
Hi, just one more question about recording.dat if you don't mind. Is it necessary to generate it? KS4 native uses the raw recording bin file, while SI-KS4 uses that recording.dat file to sort? Can SI-KS4 uses the raw recording bin file instead? If so, can I bypass the recording.dat generation?
My recording files are quite large 100-120gb, I don't think it's optimal to make another copy of the raw file in terms of time and resources.

zm711, 2025-05-10:
Another really great question (that requires a novel to explain :) ) 

Historically KS1-3 (in Matlab) required an int16 binary file organized such that it only had your raw electrophysiological data. So we had to take whatever file format users gave us and convert it to those exact specs to be used by KS1-3. With KS4 they relaxed the int16 requirement (ie they internally convert other dtypes to int16 themselves as long as the user specifies the difference) and they created an interface to use our spikeinterface extractors (which is super cool). The issue with their interface (which they note in their code directly) is that they don't use any of our multiprocessing/chunking of the extractors so going through their interface actually ends up being really really slow. So we switched back to using a default of a binary file because that allows KS4 to run much faster.

So that really brings us to your question. You have some sort of binary file so why make a `recording.dat`? So for this we need to remember that (unless KS4 has changed so feel free to link us new info if we are out of date), the binary file they are expecting is still in general meant to be a set of electrophysiological data, but most ephys recording software packs other info in the file (they typically have a header + stimulus data in the form of digital signals and ADC signals) so those types of binary files are NOT compatible directly with KS4. Instead the user needs to find someway to provide a binary file with just the ephys data. So if we look in our code 

https://github.com/SpikeInterface/spikeinterface/blob/9dec50f7f9e9026abc4bce7c4f461b53c912dbf5/src/spikeinterface/sorters/external/kilosort4.py#L143-L152

you'll see that we check if your binary file that you have provided fits the KS4 criteria and if not we write an appropriate binary file that KS4 can use (with our multiprocessing to save you time).

So that leaves you as the user of SI with a couple options. If you do:`use_binary_file=False` we will instead use the KS4-spikeinterface interface layer which will go really slow, but won't write the new binary file. Or you do `use_binary_file=True` (which is the default) in which case we write a binary file (so that's expensive for storage on your computer but will save you time in the long run) and then to clean up that storage we offer `delete_recording_dat=True` so that we only take up that extra 100GB while KS4 needs it. So you can basically decide 1) do you value your computer time more or 2) do you have limited computer resources (like storage) and so you are fine with a slow sorting to prevent extra writing to disk.

Does that all make sense? (this is so long that I'm not even going to try to proofread it sorry).

anhknguyen96, 2025-05-11:
great explanation, thank you so much! (you dont even need to proof read :) )

closed 2025-05-11 by anhknguyen96