SpikeInterface/spikeinterface #3906: Error with memmap when the run_sorter function of spikeinteface is called

Samcrl0, 2025-05-07:
Hello,
I have an issue when I use the function : 
ss.run_sorter(sorter_name="tridesclous", recording=recording_cmr, remove_existing_folder=True, output_folder = 'Sorting_'+DataName, n_jobs=1)

The entire code works very fine on another PC. However, I wanted to use another one to go faster, but it does not work anymore when it is run on the second computer. The previous function raises an error.

It seems that it cannot write at one moment in the .json file of the sorting. The mistake seems be link to the **memmap programm of numpy**. But I will not touch spontaneously to numpy...

I used for the second computer : 

- Spyder 5.5.1 with python 3.12

- neo 0.14.1
- tridesclous 1.6.8
- probinterface 0.2.28
- spikeinterface 0.102.2
- numpy 1.26.4

I do not understand why it does not work on the second computer.
Do you have an idea to solve the problem?

Thank you in advance for your help.

Error message :
``
  File c:\users\cleme\documents\traitement sam\processexpe.py:366 in Get_Sorting_and_Waveform
    self.sorting_TDC = ss.run_sorter(sorter_name="tridesclous", recording=recording_cmr, remove_existing_folder=True, output_folder = 'Sorting_'+DataName, n_jobs=1)

  File ~\anaconda3\Lib\site-packages\spikeinterface\sorters\runsorter.py:199 in run_sorter
    return run_sorter_local(**common_kwargs)

  File ~\anaconda3\Lib\site-packages\spikeinterface\sorters\runsorter.py:261 in run_sorter_local
    SorterClass.run_from_folder(folder, raise_error, verbose)

  File ~\anaconda3\Lib\site-packages\spikeinterface\sorters\basesorter.py:310 in run_from_folder
    raise SpikeSortingError(

SpikeSortingError: Spike sorting error trace:
Traceback (most recent call last):
  File "C:\Users\cleme\anaconda3\Lib\site-packages\spikeinterface\sorters\basesorter.py", line 270, in run_from_folder
    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)
  File "C:\Users\cleme\anaconda3\Lib\site-packages\spikeinterface\sorters\external\tridesclous.py", line 141, in _run_from_folder
    tdc.apply_all_catalogue_steps(cc, catalogue_nested_params, verbose=verbose)
  File "C:\Users\cleme\anaconda3\Lib\site-packages\tridesclous\cataloguetools.py", line 60, in apply_all_catalogue_steps
    cc.estimate_signals_noise(seg_num=0, duration=noise_duration)
  File "C:\Users\cleme\anaconda3\Lib\site-packages\tridesclous\catalogueconstructor.py", line 526, in estimate_signals_noise
    filtered_sigs = self.arrays.create_array(name, self.info['internal_dtype'], shape, 'memmap')
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\cleme\anaconda3\Lib\site-packages\tridesclous\iotools.py", line 143, in create_array
    arr = np.memmap(self._fname(name), dtype=dtype, mode=mode, shape=shape)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\cleme\anaconda3\Lib\site-packages\numpy\core\memmap.py", line 229, in __new__
    f_ctx = open(os_fspath(filename), ('r' if mode == 'c' else mode)+'b')
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\cleme\\Documents\\Traitement Sam\\F438_PTENSOCS3_100624_AtlasNeuro_Jul24-E32R100S1L6NT2PEDOT_ImpG_StimE0-E6-50Hz3uA1msX1Every10s_240917_185310\\SPK_240917_18h53\\Sorting_SPK_240917_18h53\\sorter_output\\channel_group_0\\catalogue_constructor\\filetered_sigs_for_noise_estimation_seg_0.raw'

Spike sorting failed. You can inspect the runtime trace in C:\Users\cleme\Documents\Traitement Sam\F438_PTENSOCS3_100624_AtlasNeuro_Jul24-E32R100S1L6NT2PEDOT_ImpG_StimE0-E6-50Hz3uA1msX1Every10s_240917_185310\SPK_240917_18h53\Sorting_SPK_240917_18h53/spikeinterface_log.json.
``

zm711, 2025-05-07:
@samuelgarcia this seems to be a TDC question. Are you using absolute paths in TDC? Maybe that is the issue.

Samcrl0, 2025-05-08:
I don't know.
What do you mean?
I touched nothing. I just download the package that I cited before, nothing else.
What should I do?

zm711, 2025-05-08:
Sorry @Samcrl0 my question was at @samuelgarcia. He wrote the TDC package so I was asking if inside the package did he use absolute paths which means that you're not able to move files around. Spikeinterface 0.102.2 should let you move stuff around without a problem, so we need to check if the problem is with tridesclous. :) 

tl;dr we just need to wait for Sam to comment.