SpikeInterface/spikeinterface #3909: set_probegroup fails when recording NWB file has sampling rate attribute instead of timestamps

dervinism, 2025-05-08:
```set_probegroup``` fails when recording NWB file has sampling rate attribute instead of timestamps, e.g.
```
recording = se.NwbRecordingExtractor(file_path=data_path, electrical_series_path='acquisition/TimeSeries_32000_Hz', \
                                     load_time_vector=True, use_pynwb=True, \
                                     cache=True, stream_cache_path=cache_path)
recording = recording.set_probegroup(probegroup)
```

It works fine when timestamps are explicitly specified when the recording NWB file is generated.

This is a shortcoming that should be fixed as NWB files can have either specification.

A repost from https://github.com/SpikeInterface/probeinterface/issues/345#issue-3047050100

dervinism, 2025-05-10:
Here it is
```
Exception has occurred: UnboundLocalError
cannot access local variable 'timestamps' where it is not associated with a value
  File "/mnt/c/Users/m329786/code_repositories/nwb2spikesort/master_file.py", line 69, in <module>
    recording = recording.set_probegroup(probegroup)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
UnboundLocalError: cannot access local variable 'timestamps' where it is not associated with a value
```

h-mayorquin, 2025-05-10:
Can you post the full one? It does not matter if it is very large.

dervinism, 2025-05-10:
Sorry, this is it
```
Traceback (most recent call last):
  File "/mnt/c/Users/m329786/code_repositories/nwb2spikesort/master_file.py", line 69, in <module>
    recording = recording.set_probegroup(probegroup)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/m329786/anaconda3/envs/si_env/lib/python3.11/site-packages/spikeinterface/core/baserecordingsnippets.py", line 103, in set_probegroup
    return self._set_probes(probegroup, group_mode=group_mode, in_place=in_place)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/m329786/anaconda3/envs/si_env/lib/python3.11/site-packages/spikeinterface/core/baserecordingsnippets.py", line 194, in _set_probes
    sub_recording = self.clone()
                    ^^^^^^^^^^^^
  File "/home/m329786/anaconda3/envs/si_env/lib/python3.11/site-packages/spikeinterface/core/base.py", line 611, in clone
    clone = BaseExtractor.from_dict(d)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/m329786/anaconda3/envs/si_env/lib/python3.11/site-packages/spikeinterface/core/base.py", line 571, in from_dict
    extractor = _load_extractor_from_dict(dictionary)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/m329786/anaconda3/envs/si_env/lib/python3.11/site-packages/spikeinterface/core/base.py", line 1127, in _load_extractor_from_dict
    extractor = extractor_class(**new_kwargs)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/m329786/anaconda3/envs/si_env/lib/python3.11/site-packages/spikeinterface/extractors/nwbextractors.py", line 590, in __init__
    ) = self._fetch_recording_segment_info_backend(file, cache, load_time_vector, samples_for_rate_estimation)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/m329786/anaconda3/envs/si_env/lib/python3.11/site-packages/spikeinterface/extractors/nwbextractors.py", line 767, in _fetch_recording_segment_info_backend
    if load_time_vector and timestamps is not None:
                            ^^^^^^^^^^
UnboundLocalError: cannot access local variable 'timestamps' where it is not associated with a value
```

h-mayorquin, 2025-05-13:
Ah, @pauladkisson already found the cause. We need to discuss the solution. Let's do this next meeting.

dervinism, 2025-05-14:
Have you found the solution? I would like to process my files as they are without resorting to regenerating them all with explicit timestamps.

h-mayorquin, 2025-05-14:
We will discuss this tomorrow. Meanwhile, I have a question: can you tell me why are you setting a probe to NWB?

dervinism, 2025-05-14:
Not sure if I understand correctly. I'm providing the probe info, so that channel coordinates are known. Isn't this info critical for spike sorting?

dervinism, 2025-05-14:
Or does the spike sorter infers coordinates from the NWB electrode table anyway?

h-mayorquin, 2025-05-14:
Yes, to better re-sate my question. 

If your NWB has `rel_x`, `rel_y` and `rel_z` in your electrodes table those values should be loaded as locations which is what you need to sort data.

Are you working on a file without that info on the electrodes table?

dervinism, 2025-05-14:
The table has columns x, y, and z. Would that work?

h-mayorquin, 2025-05-14:
When they are stored as `rel_x`, `rel_y` and `rel_z` they are loaded automatically as locations (as they represent the relative geometry of the electrodes) but it seems to me that the absolute location should work as well?

So, do your x, y and z capture the internal geometry of the probe? If yes, you can set the locations yourself which is what is actually used for the sorting.

See, you need to do something like this:

```python
recording.set_property("location", locations)
```

Which would be what set_probe does:

https://github.com/h-mayorquin/spikeinterface/blob/966cce8e65114f53e9b9eed8365af29adb480dc2/src/spikeinterface/core/baserecordingsnippets.py#L205-L211

Maybe we should load x, y and z as locations in the `NWBRecordingExtractor` if `rel_x` `rel_y` and `rel_z` are not found.

Let me know if this is clear

dervinism, 2025-05-14:
Ok, that makes and is definitely useful. Thank you. I was using the probe also to override electrode table locations if they were incorrect for some reason. This seems like a workaround.

dervinism, 2025-05-15:
I tried setting channel locations without attaching the probe like below:
```
locations = np.column_stack((electrode_table.x.values, \
                                                electrode_table.y.values, \
                                                electrode_table.z.values)) # type: ignore
recording.set_property("location", locations)
```
and ran the kilosort 4 spike sorter:
```
sorting = ss.run_sorter(sorter_name=sorter, recording=recording, # type: ignore \
                                     folder=si_output_folder_specific, \
                                     docker_image=docker_image, verbose=True, \
                                     remove_existing_folder=True, **params)
```
but got a very similar error:
```
Exception has occurred: SpikeSortingError
Spike sorting in docker failed with the following error:
Traceback (most recent call last):
  File "/mnt/c/Users/m329786/Data/spikeinterface_output/kilosort4/sorter_output/in_container_sorter_script.py", line 13, in <module>
    recording = load(json_rec)
                ^^^^^^^^^^^^^^
  File "/home/miniconda3/lib/python3.11/site-packages/spikeinterface/core/loading.py", line 100, in load
    return _load_object_from_dict(d, object_type, base_folder=base_folder)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/miniconda3/lib/python3.11/site-packages/spikeinterface/core/loading.py", line 175, in _load_object_from_dict
    return BaseExtractor.from_dict(d, base_folder=base_folder)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/miniconda3/lib/python3.11/site-packages/spikeinterface/core/base.py", line 571, in from_dict
    extractor = _load_extractor_from_dict(dictionary)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/miniconda3/lib/python3.11/site-packages/spikeinterface/core/base.py", line 1127, in _load_extractor_from_dict
    extractor = extractor_class(**new_kwargs)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/miniconda3/lib/python3.11/site-packages/spikeinterface/extractors/nwbextractors.py", line 590, in __init__
    ) = self._fetch_recording_segment_info_backend(file, cache, load_time_vector, samples_for_rate_estimation)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/miniconda3/lib/python3.11/site-packages/spikeinterface/extractors/nwbextractors.py", line 767, in _fetch_recording_segment_info_backend
    if load_time_vector and timestamps is not None:
                            ^^^^^^^^^^
UnboundLocalError: cannot access local variable 'timestamps' where it is not associated with a value
  File "/mnt/c/Users/m329786/code_repositories/nwb2spikesort/master_file.py", line 124, in <module>
    folder=si_output_folder_specific, \

                        docker_image=docker_image, verbose=True, \

                        remove_existing_folder=True, **params) # type: ignore

print(sorting)
spikeinterface.sorters.utils.misc.SpikeSortingError: Spike sorting in docker failed with the following error:
Traceback (most recent call last):
  File "/mnt/c/Users/m329786/Data/spikeinterface_output/kilosort4/sorter_output/in_container_sorter_script.py", line 13, in <module>
    recording = load(json_rec)
                ^^^^^^^^^^^^^^
  File "/home/miniconda3/lib/python3.11/site-packages/spikeinterface/core/loading.py", line 100, in load
    return _load_object_from_dict(d, object_type, base_folder=base_folder)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/miniconda3/lib/python3.11/site-packages/spikeinterface/core/loading.py", line 175, in _load_object_from_dict
    return BaseExtractor.from_dict(d, base_folder=base_folder)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/miniconda3/lib/python3.11/site-packages/spikeinterface/core/base.py", line 571, in from_dict
    extractor = _load_extractor_from_dict(dictionary)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/miniconda3/lib/python3.11/site-packages/spikeinterface/core/base.py", line 1127, in _load_extractor_from_dict
    extractor = extractor_class(**new_kwargs)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/miniconda3/lib/python3.11/site-packages/spikeinterface/extractors/nwbextractors.py", line 590, in __init__
    ) = self._fetch_recording_segment_info_backend(file, cache, load_time_vector, samples_for_rate_estimation)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/miniconda3/lib/python3.11/site-packages/spikeinterface/extractors/nwbextractors.py", line 767, in _fetch_recording_segment_info_backend
    if load_time_vector and timestamps is not None:
                            ^^^^^^^^^^
UnboundLocalError: cannot access local variable 'timestamps' where it is not associated with a value
```

h-mayorquin, 2025-05-16:
Yes, this is a bug.

Can you try?
https://github.com/SpikeInterface/spikeinterface/pull/3932

Question:
Why are you trying to using the option `load_time_vector`? how is this getting set to True?

dervinism, 2025-05-19:
That was done for data visualisation purposes. I was able to run spikesorting to completion by setting ```load_time_vector = False```. However, I cannot convert the sorter output to Phy. I get the following error:
```
Exception has occurred: ValueError
could not broadcast input array from shape (3,) into shape (2,)
  File "/mnt/c/Users/m329786/code_repositories/nwb2spikesort/master_file.py", line 138, in <module>
    "random_spikes",

    "noise_levels",

    "templates",

    "template_similarity",

    "unit_locations",

    "spike_amplitudes",

    "principal_components",

    "correlograms",

    "waveforms"

    ]

)

analyzer.compute("quality_metrics", metric_names=["snr"])
ValueError: could not broadcast input array from shape (3,) into shape (2,)
```

This is the call that produces the error:
```
analyzer = si.create_sorting_analyzer(sorting=sorting, recording=recording)
analyzer.compute([
    "random_spikes",
    "noise_levels",
    "templates",
    "template_similarity",
    "unit_locations",
    "spike_amplitudes",
    "principal_components",
    "correlograms",
    "waveforms"
    ]
)
```

dervinism, 2025-05-20:
Is anyone looking into this or should I start a separate new issue? @h-mayorquin

alejoe91, 2025-05-20:
Mmm I think that the problem is that you have 3d locations, which is not widely supported.

Can you try to do this instead?
```
locations = np.column_stack((electrode_table.x.values, \
                                                electrode_table.y.values)
recording.set_property("location", locations)
```

dervinism, 2025-05-20:
I changed to 2D coordinates and managed to run everything to completion. However, I can see that the output produced by Kilosort 4 in a standalone mode and Kilosort 4 in SpikeInterface look very different when parameters are the same. The latter looks clearly wrong but not sure what the issue is.

h-mayorquin, 2025-05-21:
Any chance that you can share your NWB file and the routines that you used with both spikeinterface and kilosort?

dervinism, 2025-05-21:
The data that gets passed to the spike sorter is for some reason garbled. NWB file has microelectrode channels all stored in int32 format. Perhaps SpikeInterface saves the intermediate binary file and then reads it assuming int16 in the process constructing garbled traces.

Here is the link to the NWB file:
https://drive.google.com/file/d/1DZVsnm71xPzUxEzga_FUN4x4JrqiUIXT/view?usp=sharing
and the probe file: 
[probe_ks4.json](https://github.com/user-attachments/files/20372467/probe_ks4.json)

You can load the json probe file and use xc and yc as 2D channel coordinates.

h-mayorquin, 2025-05-21:
Can you share the scripts that you used with both spikeinterface and kilosort? That way I can reproduce and fix your error as soon as possible.

dervinism, 2025-05-21:
Sure. This is the repo: https://github.com/dervinism/mayo2spikesort.git

Use master_file for SpikeInterface and and run_kilosort4 for standalone Kilosort. Some lines may need to be commented or uncommented.

h-mayorquin, 2025-05-21:
Does it work well when you use `save_binary=True` in the `master_file.py`?

dervinism, 2025-05-21:
I also fail to spikesort using SpykingCircus 2. I raised an issue about that https://github.com/SpikeInterface/spikeinterface/issues/3908

but there is no resolution of it either. It would be great that it could be addressed as well.

dervinism, 2025-05-21:
It works to some degree with ```save_binary=True```. It completes and the traces are not garbled. But the results are still very different from running kilosort 4 standalone. Now that you have the code and the data, perhaps you can replicate this.

dervinism, 2025-05-21:
There is also one more issue that doesn't seem to affect much of anything, but you may know better:
```
Run:
phy template-gui  /mnt/c/Users/m329786/Data/spikeinterface_output/kilosort4/phy/msel_02698_ICU03/params.py
Exception ignored in: <function _BaseNWBExtractor.__del__ at 0x7ff92bd19120>
Traceback (most recent call last):
  File "/home/m329786/anaconda3/envs/si_env/lib/python3.11/site-packages/spikeinterface/extractors/nwbextractors.py", line 432, in __del__
  File "/home/m329786/anaconda3/envs/si_env/lib/python3.11/site-packages/spikeinterface/extractors/nwbextractors.py", line 413, in _close_hdf5_file
ModuleNotFoundError: import of h5py halted; None in sys.modules
```

dervinism, 2025-05-21:
I don't mind spike sorting on ks 4 standalone. However, it would be great to be able to compare to SpykingCircus and resolving issue https://github.com/SpikeInterface/spikeinterface/issues/3908 would be really helpful.

dervinism, 2025-05-22:
I have just tried SpikeInterface with MountainSort 5 and it also fails: https://github.com/SpikeInterface/spikeinterface/issues/3948
There does not seem to be a single configuration that works. I've updated the GitHub repo, so it should be reproducible.

dervinism, 2025-05-23:
I also tried klusta and failed: https://github.com/SpikeInterface/spikeinterface/issues/3950
So far all sorters failed. Failed running: spykingcircus2, mountainsort5, klusta. Failed in outcomes: kilosort 4.

alejoe91, 2025-05-23:
Let's close this one since it's solved and focus on the other issues

closed 2025-05-23 by alejoe91