SpikeInterface/spikeinterface #3910: Missing import `get_nvidia_docker_dependecies`

ttngu207, 2025-05-09:
The recent import clean up changes - [here](https://github.com/SpikeInterface/spikeinterface/commit/f1875c16a6a2fc570e2f964d854c464a66748a81) removed the `get_nvidia_docker_dependecies` import, which is needed at this [step](https://github.com/SpikeInterface/spikeinterface/blob/8c72f55822c0c6365772251ab6931ff80300f5a6/src/spikeinterface/sorters/runsorter.py#L483)

I suspect it's a simple mistake removing the `get_nvidia_docker_dependecies` import?

alejoe91, 2025-05-09:
Oups! @zm711 maybe we can add the import locally [here](https://github.com/SpikeInterface/spikeinterface/blob/8c72f55822c0c6365772251ab6931ff80300f5a6/src/spikeinterface/sorters/runsorter.py#L482)?

zm711, 2025-05-09:
I think I know what happened. I got rid of `get_nvidia_docker_dependecies` get it is spelled wrong. But then my ID showed that the new function `get_nvidia_docker_dependencies` wasn't being used so I deleted it. Definitely my bad!

PR #3911 (2025-05-09): Add in missing `get_nvidia_docker_dependencies`
Fixes #3910
closed 2025-05-09 by alejoe91