SpikeInterface/spikeinterface #3913: Support for Neuropixels Ultra (NP1110) in SpikeInterface

Serendipityzzz, 2025-05-09:
### **Title:** Support for Neuropixels Ultra (NP1110) in SpikeInterface and DREDge

---

### **Description:**

I am using Neuropixels Ultra (NP1110) probes for neural recordings and encountered issues when processing the data using **SpikeInterface** and **DREDge**. The current implementation appears to lack explicit support for NP Ultra probes. Below are the specific issues identified and proposed modifications:

---

### **Issues Identified:**

1. **IMRO Table Parsing (SpikeInterface)**

   * The `.meta` file for NP1110 uses a different `imroTbl` format: `(channel_id, bank, mux_group)` compared to NP1/NP2.
   * Global parameters (AP/LF gains, reference, filter) are stored in the header rather than per-channel values.
   * The `_read_imro_string()` function currently expects per-channel gain/filter fields, leading to a failure when processing NP1110 data.

   **Proposed Fix:**

   * Modify `_read_imro_string()` to handle NP1110-specific parsing.
   * Extract global parameters directly from the IMRO header and adjust the data structure to use `("channel_ids", "banks", "mux_groups")`.

---

2. **Channel Index Mismatch (SpikeInterface)**

   * After slicing the probe to match `snsSaveChanSubset`, the `device_channel_indices` are reset using `np.arange()`, causing `set_probe(inplace=True)` to fail.
   * The original channel indices need to be preserved for accurate mapping.

   **Proposed Fix:**

   * Update the slicing procedure to retain original channel indices:

     ```python
     probe.set_device_channel_indices(saved_chans)  # Preserve instead of using np.arange()  
     ```

---

3. **Global Parameters Handling (SpikeInterface)**

   * NP1110 AP/LF gains and filters are defined once in the IMRO header (e.g., `(1110,2,0,500,250,1)`), while the current code assumes per-channel values.

   **Proposed Fix:**

   * Modify the `.meta` file reader to annotate the global parameters directly, e.g., `probe.annotate(global_ap_gain=500)`.

---

4. **NP Ultra Compatibility in DREDge**

   * The `read_spikeglx()` function in DREDge currently supports NP0.0, NP1.0, NP2.0, and NP1.0-NHP but not NP Ultra (1.0 HD).
   * Despite this, the demo code `ap_registration.ipynb` and the published paper demonstrate results using NP Ultra data.

   **Proposed Questions/Clarifications:**

   * How was NP Ultra compatibility enabled for DREDge in the published paper?

alejoe91, 2025-05-12:
Hi @Serendipityzzz 

Great timing for this issue! We have been collaborating with the SpikeGLX and Neuropixels developer teams to gather info about all NP probes in a centralized repo: https://github.com/billkarsh/ProbeTable

Our plan for the next couple of weeks is to update the probe parsing and the readers using that centralized resource (see https://github.com/SpikeInterface/probeinterface/issues/341). 

This said, NP1110 is the only probe within the neuropixels family that has different "rules" with respect to other probes. In the process of refactoring our parsing strategy, we'll make sure to support NP1110 correctly as well. We'll ping you here when we have a working PR for you to test.

(@jenniferColonell tagging you here if you have some additional comments!)

jenniferColonell, 2025-05-12:
Just a couple notes for @Serendipityzzz 

The NP1110 imro table is completely different than than other NP probes, due to the complicated rules for setting switches. If you need to extract any position information, please use the snsGeomMap, which records the position for each saved channel.

The data in the DREDge paper was mostly (maybe all) from a different probe (NP1100), which included only a single bank of 384 channels and no switching -- so compatibility with the NP1110 was not required.

Serendipityzzz, 2025-05-16:
@jenniferColonell Hi Jennifer! Nice to hear from you after last sfn!
The NP1100 seems to be a NP 1.0 probe, not the 1.0 HD (ultra) one (based on https://billkarsh.github.io/SpikeGLX/Sgl_help/Metadata_30.html#imec)? And in DREDge paper fig5, they show one result with NP ultra.

[image]

jenniferColonell, 2025-05-16:
Hi -- It's just a detail, but the NP1100 is an "NP1.0 like" probe, not an NP1.0 probe. The 1100 pattern is a single bank of 384 sites with the Ultra pattern. Essentially all of the data in the Ultra paper was taken with this probe type, and that's what was used in the DREDge paper as well. It is programmed just like a standard NP1.0 probe (except that there is only one bank), and the imro table is the same.

The NP1110 probe -- also called, 'Ultra', also called 'NP1.0HD' has 16 banks of 384 site each. The programming of this probe is very different, to handle that large number of sites. It therefore requires a different imro table.