SpikeInterface/spikeinterface #3920: documentation minor issues

pas-calc, 2025-05-14:
On the API documentation

- `compute_spike_amplitudes` at 
https://spikeinterface.readthedocs.io/en/stable/api.html#spikeinterface.postprocessing.compute_spike_amplitudes
has "Localize spikes in 2D or 3D with several methods given the template." but this is for `compute_spike_locations`.

- `read_kilosort` at
https://spikeinterface.readthedocs.io/en/stable/api.html#spikeinterface.extractors.read_kilosort
has "Path to the output Phy folder (containing the params.py)." but this is for `read_phy`  (however they share the same parent class)

zm711, 2025-05-14:
> compute_spike_amplitudes at
https://spikeinterface.readthedocs.io/en/stable/api.html#spikeinterface.postprocessing.compute_spike_amplitudes

Thanks yeah that was probably someone doing a bad copy :) 

> read_kilosort at
https://spikeinterface.readthedocs.io/en/stable/api.html#spikeinterface.extractors.read_kilosort
has "Path to the output Phy folder (containing the params.py)." but this is for read_phy (however they share the same parent class)

we tend to call a phy folder that set of files that ks makes that phy needs to read (it is also the set of files we make with our export to phy function). So this was intentional, but if you think it is unclear we could definitely do the Phy/Kilosort folder). Since the two projects were originally written in the same lab (but have diverged a bit) we just call the global format phy.

Any interest in submitting a PR for those fixes? Contributions are definitely welcome!

zm711, 2025-06-05:
I'll close as you've fixed both of these. Thanks again for the contributions!