SpikeInterface/spikeinterface #3921: Load SpikeGLX lf data when meta for ap is not available

h-mayorquin, 2025-05-14:
This comes from a discussion with @rly.

Currently, when loading LF, we load the probe from the AP meta:

https://github.com/SpikeInterface/spikeinterface/blob/e45a8be7fd2f65546d8dabd89e5a42fec35e3147/src/spikeinterface/extractors/neoextractors/spikeglx.py#L76-L83

The reason is that the shank map is only available in the AP meta, and that is needed to load the probe:

https://github.com/SpikeInterface/spikeinterface/pull/596#issuecomment-1130194250

I propose that if the AP meta is not present for any reason, we load the data without the probe so users can at least access it. I think this might prevent loading the inter_sample_shift property?

But maybe with all the new metadata changes in Neuropixels, we can retrieve the probe information and the inter-sample shift in a simpler way.

Thoughts?

alejoe91, 2025-05-14:
That's ok on my end!

h-mayorquin, 2025-05-14:
@samuelgarcia any thoughts? I can move forward and implement this if there are no obvious downsides. We can improve this in the future once we have better access to the metadata.

PR #3931 (2025-05-27): Read spikeglx neuropixels data without ap meta
Should fix #3921
closed 2025-05-27 by alejoe91