SpikeInterface/spikeinterface #3922: superimposing multiple individual spike waveform on the template on spikeinterface-gui and documentation about spikeinterface-gui

chiyu1203, 2025-05-15:
Dear Spikeinterface community,

After I watched the video of your [workshop](https://youtu.be/pHze_8s4Qak?si=SRmnLkg9-VhOl9F7), a thought about this function came to my mind, which is implemented already in phy. I do extracellular recording in behaving insects, which seems to have interesting waveform of spikes and MUA. Therefore, when I see atypical average waveforms, I always check if the individual waveform is a good representation of that or that is merely a byproducts of bad clustering/templates. I understand that on spikeinterface-gui, we can superimpose one spike's waveform with its unit template (as the attached picture).

[image]

However, it seems that I can not superimpose multiple spike waveform at the same time (The way Phy implement this is by "changing n spikes waveforms" on the WaveformView )

[image]

Is this something doable on spikeinterface-gui? Also I do not know how to read the x-axis and y-axis on the flattern version of Waveform View (and why there are two rows of traces).

[image]

 Lastly, I am still figuring out how to do curation on spikeinterface-gui, but I can not find instructions about that. Does anyone know where I can do some readings about these?

alejoe91, 2025-05-15:
Can you open this issue on the Spikeinterface-GUI repo? :)

chiyu1203, 2025-05-15:
good idea! [Here](https://github.com/SpikeInterface/spikeinterface-gui/issues/127) it is now

closed 2025-05-15 by chiyu1203