SpikeInterface/spikeinterface #3927: Type mismatch in upgrading to 0.102.3

mohebial, 2025-05-15:
Hi,
I've recently upgraded from version 0.101.2 to 0.102.3, and now I'm encountering a type mismatch error during motion correction (below). I'm using the same data that I previously sorted, so I know the issue isn't with my data. I would appreciate any guidance on what I might be missing. 

Thank you!,
Ali

```
 File "C:\GitHub\ePhys\spikeSorting\mna_sorting\sorting.py", line 94, in check_for_drifts
    recording_corrected, motion, motion_info = si.correct_motion(
  File "C:\Users\mna\.conda\envs\spikeSorting\lib\site-packages\spikeinterface\preprocessing\motion.py", line 557, in correct_motion
    recording_corrected = interpolate_motion(recording, motion, **interpolate_motion_kwargs)
  File "C:\Users\mna\.conda\envs\spikeSorting\lib\site-packages\spikeinterface\core\core_tools.py", line 35, in source_class_or_dict_of_sources_classes
    return source_class(*args, **kwargs)
  File "C:\Users\mna\.conda\envs\spikeSorting\lib\site-packages\spikeinterface\sortingcomponents\motion\motion_interpolation.py", line 403, in __init__
    raise ValueError(f"Can't interpolate traces of recording with non-floating dtype={recording.dtype=}.")
ValueError: Can't interpolate traces of recording with non-floating dtype=recording.dtype=dtype('int16').
```

zm711, 2025-05-15:
@alejoe91 / @samuelgarcia I don't know the workings of motion correction well enough to comment on this. I leave this for you both.

mohebial, 2025-05-15:
I didn’t convert it to float32. I confirmed that downgrading to version 0.101.2 in a separate conda environment resolves my issue. I assume that something has changed between versions, causing the correct_motion module to recognize my recording as int16. I am using the following script:
```
        recording_corrected, motion, motion_info = si.correct_motion(
            rec, 
            preset='kilosort_like', 
            folder=sorting_dir / 'motion correction', 
            output_motion=True, 
            output_motion_info=True, 
            **params.job_kwargs
        )
```

chrishalcrow, 2025-05-18:
Hello, yes we recently refactored the `correct_motion` function, and I broke the casting to floats before interpolation, very sorry. I've made a PR which fixes the bus and adds a new test so that this won't happen again.

mohebial, 2025-05-18:
Thanks, Chris!

chrishalcrow, 2025-05-19:
No worries - sorry again for the bug. Before it's merged, you'll get _extremely_ similar behaviour by casting before motion correction:

```
motion_corrected_rec = si.correct_motion(recording=recording.astype('float32'))
```

But this might increase memory usage during motion correction.

PR #3934 (2025-05-19): Bug fix: cast corrected motion recording to float
Fixes #3927

When we recently updated the `correct_motion` object, we didn't move the casting of the recording. This fixes the bug, restoring behaviour to that of the previous version.

Added a test to apply a motion correction to a `int`-typed recording.
closed 2025-05-19 by alejoe91