SpikeInterface/spikeinterface #3928: Add additional "check functions" for modules a la `available_sorters`

zm711, 2025-05-15:
Hello, I'd quite like the user to be able to get a list of available preprocessing steps, like we have for sorters: https://github.com/SpikeInterface/spikeinterface/blob/d142fe41d7b6177aea2591d68196b7cf4b6354a3/src/spikeinterface/sorters/sorterlist.py#L56 

So I'd like to expose a list, which is something like:
```
available_preprocessors = [foo.__name__ for foo in pp.preprocessor_dict.values()]
```

_Originally posted by @chrishalcrow in https://github.com/SpikeInterface/spikeinterface/pull/3898#discussion_r2075021023_

zm711, 2025-05-15:
Copying over from #3898 so we don't lose this suggestion for discussion.

chrishalcrow, 2025-05-18:
Thanks! I think it'd be great to have a unified way for the user to check what's available for them. We currently have:

`si.available_sorters()` returns the supported sorters.
`si.get_available_analyzer_extensions()` returns the supported sorting analyzer extensions

I quite like the `get` and function call. So I think it'd be great to add:

`si.get_available_extractors()` made from the many extractor dicts
`si.get_available_widgets()` made from `widget_list` in spikeinterface.widgets
`si.get_available_preprocessors()` made from `preprocessor_list`

And deprecate `available_sorters` in favour of `get_available_sorters`, then document.

I think these are cool cause they expose the user to the breadth of spikeinterface, and are helpful! Any other points in a worflow when you're stuck wondering "what options do I have??". Maybe with metrics...

zm711, 2026-10-30:
We should bring this back. Maybe as a re-ease back into a little SI coding I can work on a prototype of this @chrishalcrow .... ?