SpikeInterface/spikeinterface #3930: read_openephys could not load synchronized timestamps for OneBox

jialeren2024, 2025-05-16:
`read_openephys` using the `load_sync_timestamps=True`  can not load synchronized timestamps
but `set_times` works with the `np.load('timestamps.npy')` in the same folder

[image]

chrishalcrow, 2025-05-18:
Hi @jialeren2024 , thanks for raising this issue. What files are saved in your stream folder? As in, what's in `Record Node 101/experiment1/recording1/continuous/OneBox-100.ProbeA-AP`?

Also, what version of openephys are you using? You can find this near the top of your `Record Node 101/settings.xml` file. An example with verison 0.6.7 looks like:

```
<?xml version="1.0" encoding="UTF-8"?>

<SETTINGS>
  <INFO>
    <VERSION>0.6.7</VERSION>
```

Thanks!

zm711, 2025-05-18:
I'll also add that we don't officially support OneBox yet (if the format has differences from the base open_ephys setup. We know this is the case for the spikeglx version so I assume it is true for open_ephys to). It is on the todo list for Neo ([spikeglx-onebox issue](https://github.com/NeuralEnsemble/python-neo/issues/1684) so maybe @h-mayorquin could comment on this if he knows whether we will need to separately support OneBox too.

jialeren2024, 2025-05-19:
> Hi [@jialeren2024](https://github.com/jialeren2024) , thanks for raising this issue. What files are saved in your stream folder? As in, what's in `Record Node [...]

h-mayorquin, 2025-06-03:
We will need to add onebox support for openephys as well but we need to get testing data. At the moment, we just have testing data for spikeglx. If a user can provide some testing data for the testing repo we can move forward with the implementation.

jialeren2024, 2025-06-03:
[2025-06-03_14-42-36.zip](https://github.com/user-attachments/files/20579208/2025-06-03_14-42-36.zip)

h-mayorquin, 2025-06-03:
No, shorting is better, in fact, if you could it make shorter that would be even better for us but we can start working with this. We add data to a repository with testing examples from a lot of formats so different projects can keep testing and modifying the readers and ensuring that things keep working:

https://gin.g-node.org/NeuralEnsemble/ephy_testing_data

Can you provide a short description of the data that you shared with us?

alejoe91, 2025-06-10:
@jialeren2024 

I couldn't reproduce the timestamp issue with the short file you shared and everything seemed to have work. 

[image]

Any chance you can share the dataset that's triggerng the warning on your end?

jialeren2024, 2025-06-11:
These solved our problem for now, but if others add multiple record notes when using openephys, I think it will still be a problem.