SpikeInterface/spikeinterface #3943: Kilosort 1 customize parameters

kushaangupta, 2025-05-21:
I'm trying to run kilosort 1 with custom parameters but the default parameters that are exposed for manipulation seem only a subset of the original parameters. I do see these parameters being hard-coded here, so I was wondering if it's even possible to change these parameters

https://github.com/SpikeInterface/spikeinterface/blob/2914abdb42c4c8359493dc3401156e2cf3ea6be0/src/spikeinterface/sorters/external/kilosort.py#L134-L217

zm711, 2025-05-21:
I don't know if anyone recommends using KS1 any more (even the KS folks). There were definitely some values hardcoded, the ones that can be changed are here:

https://github.com/SpikeInterface/spikeinterface/blob/2914abdb42c4c8359493dc3401156e2cf3ea6be0/src/spikeinterface/sorters/external/kilosort.py#L37-L48

But I don't remember which of these arguments were never meant to be changed and which should be. What were you hoping to do?

kushaangupta, 2025-05-21:
I have a standardized pipeline that I want to do performance comparison with the some new sorters using spikeinterface but I cannot set all the parameters from my old pipeline to spikeinterface's KS1. The hope was to produce equivalent results and then refine with the new sorters like KS4, etc.

closed 2025-06-09 by alejoe91