SpikeInterface/spikeinterface #3947: probe_planar_contour lost when splitting & aggregating channels

RobertoDF, 2025-05-22:
Hi,
I noticed that probe inside raw_rec after the splitting and aggregating lose .probe_planar_contour.

```
        probes = read_probeinterface(probe_json)
        raw_rec = raw_rec.set_probegroup(probes, group_mode="by_shank")
        split_recording_dict = raw_rec.split_by("group")
        proc_rec = spre.highpass_filter(recording=split_recording_dict)
        proc_rec = spre.common_reference(recording=proc_rec, operator="median", reference="global")
        raw_rec = aggregate_channels(proc_rec)
        raw_rec.clear_channel_groups()

```

so I am forced to call again       `raw_rec = raw_rec.set_probegroup(probes, group_mode="by_shank")` afterward, is this fixable?

alejoe91, 2025-05-23:
Hi @RobertoDF 

I think we could check if the `probe_planar_contour` is the same for all recordings to be aggregated, and if so propagate it to the aggregated recording. Do want to give it a try?

RobertoDF, 2025-05-27:
I see, I try now