SpikeInterface/spikeinterface #3948: SpikeInterface fails with MountainSort 5: no Probe attached to recording

dervinism, 2025-05-22:
Hi. I'm having the following issue when attempting to spike sort with MountainSort 5:
```
NpzSortingExtractor: 0 units - 1 segments - 32.0kHz
  file_path: /mnt/c/Users/m329786/Data/spikeinterface_output/mountainsort5_si/sorter_output/msel_02698_ICU03/sorter_output/firings.npz
/home/m329786/anaconda3/envs/si_env/lib/python3.11/site-packages/spikeinterface/core/baserecordingsnippets.py:268: UserWarning: There is no Probe attached to this recording. Creating a dummy one with contact positions
  warn("There is no Probe attached to this recording. Creating a dummy one with contact positions")
Traceback (most recent call last):
  File "/mnt/c/Users/m329786/code_repositories/nwb2spikesort/master_file.py", line 166, in <module>
    analyzer = si.create_sorting_analyzer(sorting=sorting, recording=recording)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/m329786/anaconda3/envs/si_env/lib/python3.11/site-packages/spikeinterface/core/sortinganalyzer.py", line 152, in create_sorting_analyzer
    sparsity = estimate_sparsity(sorting, recording, **sparsity_kwargs)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/m329786/anaconda3/envs/si_env/lib/python3.11/site-packages/spikeinterface/core/sparsity.py", line 809, in estimate_sparsity
    random_spikes_indices = random_spikes_selection(
                            ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/m329786/anaconda3/envs/si_env/lib/python3.11/site-packages/spikeinterface/core/sorting_tools.py", line 217, in random_spikes_selection
    random_spikes_indices = np.concatenate(random_spikes_indices)
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: need at least one array to concatenate
Exception ignored in: <function _BaseNWBExtractor.__del__ at 0x7f1c078dd120>
Traceback (most recent call last):
  File "/home/m329786/anaconda3/envs/si_env/lib/python3.11/site-packages/spikeinterface/extractors/nwbextractors.py", line 432, in __del__
  File "/home/m329786/anaconda3/envs/si_env/lib/python3.11/site-packages/spikeinterface/extractors/nwbextractors.py", line 413, in _close_hdf5_file
ModuleNotFoundError: import of h5py halted; None in sys.modules
```

chrishalcrow, 2025-05-23:
Hi @dervinism the error actually occurs when you create the sorting analyzer, so the sorting succeeded. The error appears to be about the h5py package. What happens when you just run python and do `import h5py`? 

Also, the sorting looks like it hasn't found any units. Could you post more of your code, so we can try to figure out why that is. Thanks!

alejoe91, 2025-05-23:
@chrishalcrow but it found 0 units! :)

dervinism, 2025-05-23:
It imports h5py

dervinism, 2025-05-25:
It only finds noise clusters, that's right. Grouping electrodes allowed it to run.

alejoe91, 2025-05-26:
Let's discuss on the klusta issue then! Good luck!