SpikeInterface/spikeinterface #3950: SpikeInterface fails with klusta

dervinism, 2025-05-23:
I tried installing earlier SpikeInterface version compatible with klusta and ran it and got the error below:
```
Traceback (most recent call last):
  File "/mnt/c/Users/m329786/code_repositories/nwb2spikesort/master_file.py", line 165, in <module>
    sorting = ss.run_sorter(sorter_name=sorter, recording=recording, # type: ignore \
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/mnt/c/Users/m329786/Python_packages/spikeinterface-legacy/spikeinterface/sorters/runsorter.py", line 142, in run_sorter
    return run_sorter_local(**common_kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/mnt/c/Users/m329786/Python_packages/spikeinterface-legacy/spikeinterface/sorters/runsorter.py", line 159, in run_sorter_local
    SorterClass.set_params_to_folder(
  File "/mnt/c/Users/m329786/Python_packages/spikeinterface-legacy/spikeinterface/sorters/basesorter.py", line 167, in set_params_to_folder
    raise AttributeError('Bad parameters: ' + str(bad_params))
AttributeError: Bad parameters: ['folder']
sys:1: DeprecationWarning: Call to deprecated function (or staticmethod) _destroy.
```

```folder``` parameter was never passed to the function.

alejoe91, 2025-05-23:
Which SI version did you end up installing? So we can try to replicate

dervinism, 2025-05-23:
[0.95.1](https://github.com/SpikeInterface/spikeinterface/releases/tag/0.95.1)

alejoe91, 2025-05-23:
Can you share the entire script? The container should install the system SI version, and I suspect there is a mismatch there

dervinism, 2025-05-25:
I changed how I run it. I am getting a different error this time:
```
Traceback (most recent call last):
  File "/mnt/c/Users/m329786/code_repositories/nwb2spikesort/master_file.py", line 179, in <module>
    sorting = ss.run_sorter_by_property(sorter_name=sorter, recording=recording, # type: ignore \
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/mnt/c/Users/m329786/Python_packages/spikeinterface-legacy/spikeinterface/sorters/launcher.py", line 119, in run_sorter_by_property
    sorting_output = run_sorters([sorter_name], recording_dict, working_folder,
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/mnt/c/Users/m329786/Python_packages/spikeinterface-legacy/spikeinterface/sorters/launcher.py", line 226, in run_sorters
    assert dtype_rec_name.kind in ("i", "u", "S", "U"), "Dict keys can only be integers or strings!"
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AssertionError: Dict keys can only be integers or strings!
sys:1: DeprecationWarning: Call to deprecated function (or staticmethod) _destroy.
```
You can try to replicate the error by downloading this repo https://github.com/dervinism/mayo2spikesort.git and running master_file.py. Here is the link to download the binary: https://drive.google.com/file/d/17tyA0VQNA1gdLC88uNia0PC5wvVNx-9k/view?usp=sharing (int16 16 channels). The probe file is attached

[probe_ks4.json](https://github.com/user-attachments/files/20432811/probe_ks4.json)

alejoe91, 2025-05-26:
> I changed how I run it. I am getting a different error this time:
>
> ```
> Traceback (most recent call last):
>   File "/mnt/c/Users/m329786/code_repositories/nwb2spikesort/master_file.py", line 179, in [...]

I think you can solve this by forcing `groups` to ints [here](https://github.com/dervinism/mayo2spikesort/blob/main/master_file.py#L129) (and [here](https://github.com/dervinism/mayo2spikesort/blob/main/master_file.py#L129)):
```
recording.set_channel_groups(np.array(groups).astype(int))
```

dervinism, 2025-05-26:
Now I am getting this error:
```
/mnt/c/Users/m329786/Python_packages/spikeinterface-legacy/spikeinterface/core/baserecordingsnippets.py:211: UserWarning: There is no Probe attached to this recording. Creating a dummy one with contact positions
  warn(
write_binary_recording: 100%|################################################################################################| 1729/1729 [01:22<00:00, 21.00it/s]
RUNNING SHELL SCRIPT: /mnt/c/Users/m329786/Data/spikeinterface_output/klusta/sorter_output/msel_02750_ICU02a_micro_data/0/klusta/run_klusta.sh
NameError: name 'np' is not defined

Error running klusta
Traceback (most recent call last):
  File "/mnt/c/Users/m329786/code_repositories/nwb2spikesort/master_file.py", line 183, in <module>
    sorting = ss.run_sorter_by_property(sorter_name=sorter, recording=recording, # type: ignore \
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/mnt/c/Users/m329786/Python_packages/spikeinterface-legacy/spikeinterface/sorters/launcher.py", line 119, in run_sorter_by_property
    sorting_output = run_sorters([sorter_name], recording_dict, working_folder,
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/mnt/c/Users/m329786/Python_packages/spikeinterface-legacy/spikeinterface/sorters/launcher.py", line 269, in run_sorters
    _run_one(task_args)
  File "/mnt/c/Users/m329786/Python_packages/spikeinterface-legacy/spikeinterface/sorters/launcher.py", line 38, in _run_one
    run_sorter(sorter_name, recording, output_folder=output_folder,
  File "/mnt/c/Users/m329786/Python_packages/spikeinterface-legacy/spikeinterface/sorters/runsorter.py", line 142, in run_sorter
    return run_sorter_local(**common_kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/mnt/c/Users/m329786/Python_packages/spikeinterface-legacy/spikeinterface/sorters/runsorter.py", line 164, in run_sorter_local
    sorting = SorterClass.get_result_from_folder(output_folder)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/mnt/c/Users/m329786/Python_packages/spikeinterface-legacy/spikeinterface/sorters/basesorter.py", line 278, in get_result_from_folder
    raise SpikeSortingError(
spikeinterface.sorters.utils.misc.SpikeSortingError: Spike sorting failed. You can inspect the runtime trace in spikeinterface_log.json
sys:1: DeprecationWarning: Call to deprecated function (or staticmethod) _destroy.
```

alejoe91, 2025-05-26:
Mmm...not really sure why this is happening and klusta is deprecated.. Since you're using a different SI version for it, maybe the easiest way to run it is to install klusta in the same env (should be Python3.7).

Then you just run:
```
pip install Cython h5py tqdm six scipy
pip install click klusta klustakwik2==0.2.7
```
and klusta should be installed. Unfortunately given that klusta needs a very old version of SpikeInterface it would make it quite hard to debug it on our end..

alejoe91, 2025-05-26:
I think this comes from VS-code. Try to create the env and install all dependencies from the command line instead

dervinism, 2025-05-26:
Runing directly in the terminal, but getting the following error:
```
RUNNING SHELL SCRIPT: /mnt/c/Users/m329786/Data/spikeinterface_output/klusta/sorter_output/msel_02750_ICU02a_micro_data/0/klusta/run_klusta.sh
11:44:24 [I] creator:446          Using kwik file `/mnt/c/Users/m329786/Data/spikeinterface_output/klusta/sorter_output/msel_02750_ICU02a_micro_data/0/klusta/recording.kwik`.

11:44:24 [I] launch:206           Starting spike detection.

11:44:24 [I] launch:53            SpikeDetekt parameters:

11:44:24 [I] launch:55            chunk_overlap_seconds = 0.015

11:44:24 [I] launch:55            chunk_size_seconds = 3

11:44:24 [I] launch:55            connected_component_join_size = 1

11:44:24 [I] launch:55            detect_spikes = both

11:44:24 [I] launch:55            dtype = int16

11:44:24 [I] launch:55            excerpt_size_seconds = 1

11:44:24 [I] launch:55            experiment_name = recording.kwik

11:44:24 [I] launch:55            extract_s_after = 43

11:44:24 [I] launch:55            extract_s_before = 21

11:44:24 [I] launch:55            filter_butter_order = 3

11:44:24 [I] launch:55            filter_high_factor = 0.475

11:44:24 [I] launch:55            filter_low = 500.0

11:44:24 [I] launch:55            n_channels = 4

11:44:24 [I] launch:55            n_excerpts = 50

11:44:24 [I] launch:55            n_features_per_channel = 3

11:44:24 [I] launch:55            pca_n_waveforms_max = 10000

11:44:24 [I] launch:55            prb_file = /mnt/c/Users/m329786/Data/spikeinterface_output/klusta/sorter_output/msel_02750_ICU02a_micro_data/0/klusta/probe.prb

11:44:24 [I] launch:55            raw_data_files = /mnt/c/Users/m329786/Data/spikeinterface_output/klusta/sorter_output/msel_02750_ICU02a_micro_data/0/klusta/recording.dat

11:44:24 [I] launch:55            sample_rate = 32000.0

11:44:24 [I] launch:55            threshold_strong_std_factor = 5

11:44:24 [I] launch:55            threshold_weak_std_factor = 2

11:44:24 [I] launch:55            use_single_threshold = 1

11:44:24 [I] launch:55            voltage_gain = 1.0

11:44:24 [I] launch:55            weight_power = 2

11:44:24 [I] spikedetekt:211      Found 4 live channels in group 0.

11:44:24 [I] spikedetekt:285      Finding the thresholds...

Detecting spikes        :   0%|          | 0/580 [00:00<?, ?it/s]/home/m329786/.local/lib/python3.12/site-packages/klusta/traces/store.py:100: FutureWarning: In the future `np.object` will be defined as the corresponding NumPy scalar.

  assert dtype != np.object

Detecting spikes        :   0%|          | 0/580 [00:00<?, ?it/s]

AttributeError: module 'numpy' has no attribute 'object'.

`np.object` was a deprecated alias for the builtin `object`. To avoid this error in existing code, use `object` by itself. Doing this will not modify any behavior and is safe.

The aliases was originally deprecated in NumPy 1.20; for more details and guidance see the original release note at:

    https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations

Error running klusta
Traceback (most recent call last):
  File "master_file.py", line 182, in <module>
    grouping_property='group', **params) # type: ignore
  File "/mnt/c/Users/m329786/Python_packages/spikeinterface-legacy/spikeinterface/sorters/launcher.py", line 127, in run_sorter_by_property
    sorter_params={sorter_name: sorter_params})
  File "/mnt/c/Users/m329786/Python_packages/spikeinterface-legacy/spikeinterface/sorters/launcher.py", line 269, in run_sorters
    _run_one(task_args)
  File "/mnt/c/Users/m329786/Python_packages/spikeinterface-legacy/spikeinterface/sorters/launcher.py", line 43, in _run_one
    with_output=with_output, **sorter_params)
  File "/mnt/c/Users/m329786/Python_packages/spikeinterface-legacy/spikeinterface/sorters/runsorter.py", line 142, in run_sorter
    return run_sorter_local(**common_kwargs)
  File "/mnt/c/Users/m329786/Python_packages/spikeinterface-legacy/spikeinterface/sorters/runsorter.py", line 164, in run_sorter_local
    sorting = SorterClass.get_result_from_folder(output_folder)
  File "/mnt/c/Users/m329786/Python_packages/spikeinterface-legacy/spikeinterface/sorters/basesorter.py", line 279, in get_result_from_folder
    "Spike sorting failed. You can inspect the runtime trace in spikeinterface_log.json")
spikeinterface.sorters.utils.misc.SpikeSortingError: Spike sorting failed. You can inspect the runtime trace in spikeinterface_log.json
```

dervinism, 2025-05-27:
I also tried ```numpy==1.20``` and ```numpy==1.19```, but still getting the same error

alejoe91, 2025-05-27:
not sure how to fix it then :( it's weird that you still get the error though because it says that deprecation started in v1.20, so it should not raise an error!

dervinism, 2025-05-27:
Yes, the numpy is 1.19. So I guess this will remain a mystery. I replaced all the instances of ```np.object``` to ```object``` in ```store.py``` and ```spikedetekt.py``` and I was able to complete the spike sorting. However, now I'm stuck on converting results to Phy. I get the error below:
```
klusta run time 183.78s
/mnt/c/Users/m329786/Python_packages/spikeinterface-legacy/spikeinterface/core/unitsaggregationsorting.py:72: UserWarning: Cannot propagate registered recording to UnitsAggregationSorting
  "Cannot propagate registered recording to UnitsAggregationSorting")
UnitsAggregationSorting: 62 units - 1 segments - 32.0kHz
Traceback (most recent call last):
  File "master_file.py", line 188, in <module>
    analyzer = si.create_sorting_analyzer(sorting=sorting, recording=recording)
AttributeError: module 'spikeinterface.full' has no attribute 'create_sorting_analyzer'
```
What should I be using instead of ```create_sorting_analyzer```?

alejoe91, 2025-05-27:
You should save the `sorting` object (with `sorting.save(folder="...")`, then reload it with the latest version `sorting = si.load(folder=...)`

closed 2025-05-29 by alejoe91