SpikeInterface/spikeinterface #3951: Kilosort 4 for Neuropixels 2

harshk95, 2025-05-23:
Hi, 
I have tried to look for a concerted tutorial for implementing KS4 for NPX2.0 probes but I might just be missing it in the docs. Is there a compiled document/notebook for the entire process readily availble: from the preprocessing into groups and then sorting and saving the aggregated recording and the associated channel maps (after changing the sort order)
Sorry if this is redundant I could not readily find the related information.

Thanks a lot!

zm711, 2025-05-26:
Thanks @chiyu1203 approach you linking the docs. @harshk95 we try to stay as agnostic as possible so we tend to include internal sorters in docs. The link to the neuropixel tutorial is the other important for doing things that preprocessing in the np context. Let us know if you have more questions though.

chrishalcrow, 2025-08-19:
Hello, at the moment `get_noise_levels` only accepts Recordings (we could modify it so it accepts dicts too, I'm not sure how helpful this is). So you can do something like:

``` python
noise_levels = {}
for key, recording in good_channels_recording.items():
    noise_levels[key] = si.get_noise_levels(recording)
```

or the aggregation that you show should work. Could you try doing it the way I've shown above: that will hopefully show if it's a problem with the aggregation or not.