SpikeInterface/spikeinterface #3963: NPX2.0 Multi shank - sort by group with KS4: discrepancy in channel number and channel group assignment when exported

harshk95, 2025-05-30:
Hi, 
I have just noticed an issue when I sort an NPX2.0 recording with KS4, with sort by group. In phy while checking the units I noticed that some of the channels had units with different channel groups assigned to the same channel. The ones where the channel number matches the real channel group (shank = 3) the raw waveform traces look good whereas for the units assigned to that channel with different channel group numbers, the waveform view is flat for those channels with very low amplitude suggesting that these units seem to be wrongly assigned to that channel. See the screenshot below: 

[image]

One can see that all these units have the same channel but all different channel groups and the ones on the 'correct' shank have reasonable amplitude values. 

[image]
This is a unit assigned in the correct channel group (3).

[image]
This above one instead is representative of the ones in the 'wrong' channel groups.

I am using 
KS 4.0.37
SpikeInterface 0.102.4
Windows 10, NVIDIA RTX 3080 Ti

Below is the respective code used in the pipeline:

`
splitRec = raw_rec.split_by(property='group')

recs = {}
for group, sub_recording in tqdm(splitRec.items()):
    sub_recording = si.depth_order(sub_recording)
    rec = si.bandpass_filter(sub_recording, freq_min=300., freq_max=6000.)
    bad_channel_ids, channel_labels = si.detect_bad_channels(rec, seed = 1)
    rec = si.interpolate_bad_channels(recording=rec, bad_channel_ids=bad_channel_ids)
    print('Number of noisy channels:', len(bad_channel_ids))
    rec = si.remove_artifacts(rec, laser_events, ms_before=0.5, ms_after=0.5)
    rec = si.phase_shift(rec)
    rec = si.common_reference(rec, operator="median", reference="global")
    ## Additionally destripe to see impact
    rec = si.highpass_spatial_filter(recording=rec)

    recaux = si.bandpass_filter(sub_recording, freq_min=300., freq_max=5000.)
    recaux = si.phase_shift(recaux)
    recaux = si.common_reference(recaux, operator="median", reference="global")
    motionFolder = join(imec_folder, f'drift_info_{group}')
    rec_corrected_aux, motion, motion_info  = si.correct_motion(recording=recaux, preset="dredge", output_motion_info  = True,output_motion=True, folder = motionFolder, overwrite=True, estimate_motion_kwargs={'win_step_um':50,'win_scale_um':50}, **job_kwargs)
    motion_info = si.load_motion_info(motionFolder)
    rec = si.astype(rec, np.float32)
    rec_corrected = interpolate_motion(recording=rec, motion=motion_info['motion']) # **motion_info['parameters']['interpolate_motion_kwargs'])
    recs[group] = rec_corrected
    
    fig = plt.figure(figsize=(14, 8))
    si.plot_motion_info(
        motion_info, recaux,
        figure=fig,
        #depth_lim=(400, 600),
        color_amplitude=True,
        amplitude_cmap="inferno",
        scatter_decimate=5,
    )
    plt.savefig(join(motionFolder, 'Drift estimates.png'))
    plt.savefig(join(motionFolder, 'Drift estimates.pdf'))
    
    plt.show()
    plt.close()

combined_recording = si.aggregate_channels(list(recs.values()))

`

These are the preprocessing steps. Next is the sorting:

`
params_kilosort4 = {'nblocks': 5, 'do_correction':False}
sortingKS4 = si.run_sorter_by_property('kilosort4',
                                       combined_recording,
                                       grouping_property='group', 
                                       folder=ks_int_dir,
                                       remove_existing_folder=True,
                                       delete_output_folder=False,
                                       verbose=True, 
                                       engine='loop', #..engine_kwargs={"n_jobs": 16}
                                       torch_device='cuda', **params_kilosort4)

`

And finally export to phy:

`
sortClean =si.remove_duplicated_spikes(sorting = sortingKS4)
analyzerKS4 = si.create_sorting_analyzer(sortClean, combined_recording, sparse=True, format="memory")
analyzerKS4.compute(["random_spikes", "waveforms", "templates", "noise_levels","correlograms","unit_locations", "template_similarity"], **job_kwargs)
sortClean = si.remove_redundant_units(sorting_or_sorting_analyzer = analyzerKS4)
analyzerKS4 = si.create_sorting_analyzer(sortClean, combined_recording, sparse=True, format="memory")
analyzerKS4.compute(["random_spikes", "waveforms", "templates", "noise_levels","correlograms","unit_locations", "template_similarity"], **job_kwargs)

metric_names=['firing_rate', 'firing_rate', 'presence_ratio', 'snr', 'isi_violation', 'amplitude_cutoff', ]
metricsKS4 = si.compute_quality_metrics(analyzerKS4, metric_names=metric_names)

si.export_to_phy(analyzer_cleanKS4,kilosort_folder ,copy_binary=True, compute_pc_features=False, remove_if_exists = False, verbose=True, n_jobs=-1)

`

I have not tried to extensively troubleshoot the issue and wanted to check if I am doing something wrong or have a gap in my understanding of the sorting by group. Any help on this matter would be appreciated.

Thank you!

harshk95, 2025-06-02:
So, some updates,
I ran it without preprocessing and postprocessing with the minimal code as @chrishalcrow suggested, and seems like I still get the weird spread of units that seem to be wrongly assigned and still have the weird lack of spikes at the aligned timestamps.

[image]

[image]

The other 'weird' thing is that for this same sort I had reported that the sorting was very slow ( #3591  ). Now excluding the time for the preprocessing, the sorting was 4x times faster even though I enabled the kilosort_correction.

chrishalcrow, 2025-06-02:
Interesting @harshk95 !

I think the fact that the simple sorting still found these weird units suggests that you have some weird noise on your probe, rather than sorting problem. Maybe someone with more experience can chip in (@alejoe91 ??). If this is the case: you can either try to clean up the recording (hard!) or curate the units (maybe less hard).

The `spike_locations` computation assumes the waveform has a pretty standard form. When they are noisy, the assumption breaks down and you get inaccurate locations (because noise doesn't really have a central location!).

Good to know about the timing - we need to do a proper study of these types of things...

alejoe91, 2025-06-02:
Hey guys, so the weird thing is that the correlogram looks very clean..

Anyways, @harshk95 we have a sparsity mode called `by_propety`. Since you are sorting by property "group", the units will get assigned a group property as well and you can use this mode, which will use the channels of the shank a unit was found on as sparsity. Let's see if this clears things up a bit

alejoe91, 2025-06-02:
Another comment: usually when you see these unit clusters, it's mainly all noise chopped up in many small units...for the good units, do you see a matches between unit templates and locations? If that's the case, then I would say not to spend too much energy on debugging "noise" ;)

harshk95, 2025-06-06:
The sparsity mode by property seems to have the trick, maybe the find unit_locations was doign something weird with the existing sparse = True alone. 
The main thing was it does now seem like some of those units must have been real and just wanted to be sure we were not systematically excluding something or reading the channel maps wrong.

Thakns for the help!

P.S. We wanted to know a bit about how to use the unit locations after registering the probe via the IBL GUI for example and if there was an easy way to extract the channel corresponding to the unit easily with an existing mehtod. I will make a new issue about this anyway.

closed 2025-06-06 by harshk95