SpikeInterface/spikeinterface #3964: reading Binary recording files

lihao881230, 2025-05-31:
Hello, does anyone know how to read recordings in .bin files? We use the WhiteMatter acquisition system that records using the OpenEphys GUI. 

[image]

zm711, 2025-06-01:
We are working on a white matter recording extractor. The pilot version is [here](https://github.com/SpikeInterface/spikeinterface/blob/main/src/spikeinterface/extractors/whitematterrecordingextractor.py) cc @pauladkisson ? Any comments here for white matter? 

Not sure if the openephys guy transforms the data. We also have an openephy reader too which can be obtained with `from spikeinterface.extractors import read_ophenephys`. If that doesn't work we also allow people to load a binary file themself using

`from spikeinterface.extractors import read_binary`

But in this case you need to supply `dtype`, `n_channels`, and `sampling_frequency`.

zm711, 2025-06-19:
@lihao881230,

Could you provide us any additional info. We want to fully support `WhiteMatter` so a spec and test files/folder would be amazing for us.

h-mayorquin, 2025-06-19:
Hi, the most important for us is to know if you have the spec.

Which will answer these questions:

> Hi, are the events also produced by the WhiteMatter? Are the signals multiplexed? Do you have access to the spec for the analog/events format?

h-mayorquin, 2025-06-19:
Hi, thanks, actually the settings.xml do have a lot of info that we did not know was available for this format (such as the gains and the binary layout of the analog and digital pannels).

I was not clear but when I was asking for the spec I was asking if you did not have a document (pdf or other format) where the format is described. This is usually provided by the manufacturer of the acquisition system.

Also, I don't see the notebook in the files that you shared.

lihao881230, 2025-06-19:
Hi, here are the notebooks. Please let me know if anything else is unclear. I really appreciate your help and time to getting this standardized for the whitematter systems.Hao

h-mayorquin, 2025-06-19:
Hey, I think that you are answering things on an email and probably attaching files but you actually need to access github and share the file here if you want us to be able to see it.

lihao881230, 2025-06-23:
> Thanks a lot. I think we can come back here:
>
> Do you happen to have an extractor available for the analog event channels
> as well (also in .bin format)? Or would it be best for me to load the event
> data manually using read_binary and compute the relevant event times myself?
>
> I got the notebook and it looks like you're successfully loading the analog
> data using the extractor, which should work fine.
>
> Just a couple of things to point out:
>
> One issue is the sampling frequency. From the metadata files, it looks like
> the sampling rate is 25,000 Hz, but in your notebook you're using 30,000.
> Why is that?
>
> The other issue is the gain. Right now the extractor assumes a default gain
> that works for the headstages:
>
> https://github.com/SpikeInterface/spikeinterface/blob/2afae6077c2a0cb9fe6fd6cf8e98d09fcb921389/src/spikeinterface/extractors/whitematterrecordingextractor.py#L33-L34
>
> But this assumption doesn’t apply to the analog panel channels. According
> to the metadata that I saw, the gains are different for the headstages. so
> using the default would scale your signals incorrectly by a large factor. I
> think that you are accounting for it on the threshold but just something to
> think on.
> [... 23 lines cut ...]
> segmentation or chunk timing, so I'm wondering how you know they’re
> contiguous. It seems to be the case from the timestamps, but I couldn’t
> find anything that documents it. Is that something that you set on the
> configuration of the system? I am asking because maybe we can concat
> automatically if there is a way of detecting this.

Hi, thank you for reviewing my code and catching these errors! And yes,
please feel free to use any of the data I uploaded. If necessary, we can
collect additional data for this.

The segmentation duration can be set when we initiate the
acquisition software, and it just happened to be set as 5 mins for these
recordings. I don't know if it saves a log somewhere other than the file
names, but I can check with the company on that. And I believe they are
continuous.