SpikeInterface/spikeinterface #3965: Pytest 8.4.0 breaking core test build summary

zm711, 2025-06-02:
Pytest 8.4.0 just released today and now the core test build summary is broken. I haven't found what in the changelog has been altered to lead to this break. copying error below for us:
```
Successfully installed pandas-2.2.3 python-dateutil-2.9.0.post0 pytz-2025.2 six-1.17.0 tzdata-2025.2
Collecting tabulate
  Downloading tabulate-0.9.0-py3-none-any.whl.metadata (34 kB)
Downloading tabulate-0.9.0-py3-none-any.whl (35 kB)
Installing collected packages: tabulate
Successfully installed tabulate-0.9.0
Traceback (most recent call last):
  File "/Users/runner/work/spikeinterface/spikeinterface/./.github/scripts/build_job_summary.py", line 26, in <module>
    timing_column = [float(line.split("s")[0].rstrip()) for line in timing_info]
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/runner/work/spikeinterface/spikeinterface/./.github/scripts/build_job_summary.py", line 26, in <listcomp>
    timing_column = [float(line.split("s")[0].rstrip()) for line in timing_info]
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: could not convert string to float: ''
```
@h-mayorquin did you write this summary? Or was it someone else? We could always limit pytest for now. I'll open a PR, but if you want to make a more permanent solution that would be good.

h-mayorquin, 2025-06-02:
I wrote the summary, I can take a look but let's pin the test in the meanwhile. It is not like we want to use any of the new features as soon as possible.

zm711, 2025-06-03:
I'll leave this open while we wait for a long-term fix. We can close when we release the pytest limit.