SpikeInterface/spikeinterface #3969: resample() converts dtype to int16 even when dtype=‘uint16’ is specified

carmengasco, 2025-06-03:
Hello!

When I load a binary file using read_binary, I specify dtype='uint16'. As expected, the trace I get is indeed of type uint16.

However, the issue arises when I use .resample(recording, 2500, dtype='uint16'). Even though I explicitly set the output dtype, the resulting data is automatically converted to int16.

Is there a way to preserve the original uint16 dtype during resampling?

Thanks a lot!

carmengasco, 2025-06-03:
This is the resample method: 
https://spikeinterface.readthedocs.io/en/stable/api.html#spikeinterface.preprocessing.resample

[image]

h-mayorquin, 2025-06-03:
I think this is for @alejoe91 and @samuelgarcia

zm711, 2025-07-03:
Since we moved forward with #3982 this is true for all preprocessing and a warning will be raised explaining issues with with using unsigned int's within the code base so I'll close this. But if you have more questions just let us know!

closed 2025-07-03 by zm711