SpikeInterface/spikeinterface #3997: check_probe_do_not_overlap only checks x and y

AOONIH, 2025-06-18:
Hi,
I am using a double sided probe so my coordianates are in 3D. I make sure to sort the probe by group, however, I run into the issue when trying to load the recording with the 3D probegroup as check_probe_do_not_overlap raises the exception "Exception: Probes are overlapping! Retrieve locations of single probes separately". Further inspetion shows that the function only checks the first 2 dimensions and not the 3rd for a 3D probe.

Is there away to bypass this when reading in the recordings?

AOONIH, 2025-06-18:
I'm using this probe from cambridge neurotech: https://www.cambridgeneurotech.com/assets/files/ASSY-325D-P-1_P-2-map.pdf
To generate the pribe I create a probegroup where the sites on the 2 sides are offset by 30 um in the z dim and rotated (this is more of a probeinterface topic). 
Regarding the chek overlapping func, I think the function should use as all the probe dimensions to determining whether the site are overlapping and not just the first 2.

I slightly modified the function and can now load the recording:
    def check_probe_do_not_overlap(probes):
        """
        Ensure that multiple probes do not have overlapping contact positions 
        in N-dimensional space by using axis-aligned bounding boxes.

        Parameters
        ----------
        probes : list
            A list of probes, each with a `contact_positions` attribute 
            as a NumPy array of shape (n_contacts, n_dims)

        Raises
        ------
        Exception
            If any probes have overlapping contact positions
        """
        for i in range(len(probes)):
            probe_i = probes[i]
            positions_i = probe_i.contact_positions
            min_i = np.min(positions_i, axis=0)  # (n_dims,)
            max_i = np.max(positions_i, axis=0)  # (n_dims,)

            for j in range(i + 1, len(probes)):
                probe_j = probes[j]
                positions_j = probe_j.contact_positions  # shape (n_contacts, n_dims)

                # Check if each contact in probe_j is inside the bounding box of probe_i
                in_bounds = np.all((positions_j >= min_i) & (positions_j <= max_i), axis=1)
                if np.any(in_bounds):
                    raise Exception(
                        f"Probes {i} and {j} are overlapping! Retrieve locations of single probes separately.")
As an aside, I thought trisdeclous or mountatinsort5 would allow for 3D positions. I don't know if there is any specific workflow for this

samuelgarcia, 2025-06-19:
Hi Dammy.
Zach is right no sorter handle correctly the double side probes geometry
And it even worst : probeinterface do not handle it! If you keep contact at the same position you will have the warning you already experience.
The only way is to have the B side virtualy in the same plan and far away enough from the A side.
The Z axis is really poorly handled. Sorry for that.