SpikeInterface/spikeinterface #3998: Issue using docker container with Kilosort 2 and 2.5 : error while loading shared libraries

jmcdaniel2, 2025-06-18:
Hi all,
When I am trying to use kilosort 2 or 2.5 through the use of a docker container I get the following error 

`"ks2_5_compiled: error while loading shared libraries: /opt/matlabruntime/R2022b/sys/os/glnxa64/libstdc++.so.6: file too short"`

The code I am using to run the sorter is 

```
import spikeinterface.full as si

basepath = Path(r"D:\Ephys\20250611_LY_TS232F\Record Node 101\experiment1")
rec3 = si.read_openephys(basepath / 'recording3',stream_name='Record Node 101#OneBox-100.ProbeA-AP') 

params = si.get_default_sorter_params('kilosort2_5')
sorting_KS2 = si.run_sorter('kilosort2_5',rec3, folder = basepath / 'rec3kilosort2_5', verbose = True, docker_image = True,installation_mode = 'pypi',**params)
```
Any help figuring this out would be greatly appreciated!

samuelgarcia, 2025-06-19:
Oups. This looks like a deep and painfull bug in the built container.
@alejoe91 : do you still have residual of power to handle this docker ks2 stuff ?

alejoe91, 2025-06-19:
Can you share the full trace?

@samuelgarcia I plan to rebuild the matlab containers with a newer Python version sometimes next week