SpikeInterface/spikeinterface #4002: Error when trying to sort with kilosort 2.5

edisontom1, 2025-06-21:
Hi,

I was running into an issue that I could not find anywhere else with tuple index out of range. Would this be a problem with my channel map? My channel map is a numpy array of size 64 and the dataset contains 64 channels recorded from two probes with the same layout. I have attached the issue and my code below. Thanks!

[image]

[image]

zm711, 2025-06-26:
Hey @edisontom1,

could you post how you are making `chanMap` we would need to see why there is a problem there :)

edisontom1, 2025-06-27:
Would my chanMap be incorrect if this is what prints out when I run the get channel id function on my recording?

['1' '2' '3' '4' '5' '6' '7' '8' '9' '10' '11' '12' '13' '14' '15' '16'
 '17' '18' '19' '20' '21' '22' '23' '24' '25' '26' '27' '28' '29' '30'
 '31' '32' '65' '66' '67' '68' '69' '70' '71' '72' '73' '74' '75' '76'
 '77' '78' '79' '80' '81' '82' '83' '84' '85' '86' '87' '88' '89' '90'
 '91' '92' '93' '94' '95' '96']

alejoe91, 2025-06-28:
> Hi, 
> chanMap is a numpy array of size 64 that I hardcoded according to this image. I tried messing with the size and it seemed to not be the issue as when I tried using a different sized array it gave me a different error. Would it be an issue with any of the other parameters? 
> 
> [image]
> 
> chanMap = (np.array([18, 16, 20,14,17,15,19,13,31,1,29,3,32,2,27,5,30,4,25,7,28,6,23,9,26,8,21,11,24,10,22,12] + [17+32,16+32,18+32,15+32,19+32,14+32,20+32,13+32,21+32,12+32,22+32,11+32,23+32,10+32,24+32,9+32,25+32,8+32,26+32,7+32,27+32,6+32,28+32,5+32,29+32,4+32,30+32,3+32,31+32,2+32,32+32,1+32]) -1)

Isn't it this probe? The channel locations are the x-y position of each electrode. You can get them from the probe spec

edisontom1, 2025-06-28:
I see so am I creating the numpy array wrong? For example would the first element of my new array for setting the channel location be [18,17]? Also I was just curious how I would handle two of the same probe and would I just need to match up the 2nd 32 channels with the values in the channel id?