SpikeInterface/spikeinterface #4003: Extract all spike waveforms without random subsampling in extract_waveforms

mgemayel, 2025-06-22:
Hello,

I am using the extract_waveforms function from spikeinterface.core to extract spike waveforms for further analysis. I understand from the documentation that random subsampling occurs during waveform extraction through the max_spikes_per_unit parameter.

For my analysis, I would like to extract waveforms for all spikes without any subsampling. I have tried setting max_spikes_per_unit to a very high number, but it still appears to randomly subsample.

Could you please advise on the best way to achieve this? 

Thank you very much for your help,
Mia

samuelgarcia, 2025-06-23:
Hi,
`extract_waveforms()` is deprecated and has been replace by the concept of Sorting analyzer.
See https://spikeinterface.readthedocs.io/en/stable/modules/postprocessing.html.

In the analyzer the extension "random_spikes" internaly use the `random_spikes_selection()` that `handle methpd="all"`that takes all your spikes. Do not this for high channel count with no sparsity, the waveforms would be bigger than the dataset itself!!

Note that you also have a low level fonction `extract_waveforms_to_single_buffer()` to only run the snipet extraction.

zm711, 2025-07-17:
As this answers the question I will close this for now. @mgemayel if this doesn't work for you or you have more questions feel free to reopen or open a new issue. We love to help people!

closed 2025-07-17 by zm711