SpikeInterface/spikeinterface #4007: Issue: Discrepancy Results Between SpikeInterface (using kilosort4 as sorter) and Standalone KiloSort4

darianyao, 2025-06-23:
Hi, dear spikeinterface developer.

I'm encountering a significant discrepancy between spikeinterface and kilosort 4 when using kilosort 4 example data as dataset.   I first preprocess the data using spikeinterface (applying filtering and Common Median Re-referencing (CMR)), and then use spikeinterface.sorters.kilosort4 to sort the preprocessed data. However, the results from this pipeline differ substantially from those obtained by directly running KiloSort4 on the raw example data without any prior preprocessing in SpikeInterface.

Could this difference in results be due to KiloSort4 performing its own internal preprocessing steps, effectively leading to the data being preprocessed twice when using the SpikeInterface pipeline?

And could you please give me some suggestions when using KiloSort4 as a sorter, is it still necessary to perform data preprocessing with SpikeInterface beforehand? If not, what data should be used for calculating extensions when performing quality metrics calculations?

Thanks a lot!

chrishalcrow, 2025-06-23:
Hello, there's quite a lot of discussion about this topic here: #3483. I think my comment here might be helpful: https://github.com/SpikeInterface/spikeinterface/issues/3483#issuecomment-2859188115

alejoe91, 2025-06-24:
> Thanks for your help! Your previous advice was very helpful.
> 
> I'm now trying to apply it to my own data, but I've run into a new problem. I'm getting an error that the module 'spikeinterface.full' has no attribute 'interpolate_motion', even though my installed SpikeInterface version is 0.102.3.

The `interpolate_motion` is not exposed at the full level. I think that @chrishalcrow's comment was more qualitative than fully functional :)

You can either do:
```
from spikeinterface.sortingcomponents.motion import interpolate_motion
```

Or (better if you want to compute and interpolate):
```
import spikeinterface.preprocessing as spre

# this does compute+interpolate
recording_corrected = spre.correct_motion(recording, preset=...)
```

chrishalcrow, 2025-06-25:
Hello, sorry, I'm not an expert on this! But my understanding it that to mimic kilosort, you need to whiten before computing motion, like my code snippet you share. This computes the motion of the probe during the recording. I think that _all_ motion correction algorithms assume that there is whitened data. 

But then you can apply the motion object you obtain to a non-whitened recording. In the code snippet `motion_corrected_recording` is not whitened

(sorry for the bug, you need to run
``` python
from spikeinterface.sortingcomponents.motion import interpolate_motion
motion_corrected_recording = interpolate_motion(si.common_reference(si.bandpass_filter(recording)), motion=motion)
```
)

Then when *I* make an analyzer, I like to see the waveforms etc in physical units (rather than whitened units). So I want to give the analyzer the non-whitened `motion_corrected_recording`.

I hope that helps! Other people might have some other advice!

JoeZiminski, 2025-06-26:
Hey @darianyao, I am curious that the results were different from running in spikeinterface with minimal preprocessing vs. kilosort. Could you possibly share the code? 

For kilosort, the [order of preprocessing](https://github.com/MouseLand/Kilosort/blob/edde277eacbf699560658c98c3552bff143d23cb/kilosort/run_kilosort.py#L266) is:
1. highpass filter (300 Hz by default)
2. common average reference
3. whiten
4. motion correction (kilosort method)

However, you can do the first part in spikeinterface:
1. phase shift (this is useful to do but shouldn't have a huge affect)
2. bandpass filter (say 300Hz, 6000Hz)
3. common median reference
4. <pass to kilosort>

Because the filtering and referencing is done on the SI side, you can turn this off in kilosort. You can pass `do_CAR=False` and `highpass_cutoff=100` to `run_sorter()`. This turns off kilosort CAR and sets the highpass filter to a cutoff lower than the one used in the spikieinterface bandpass (because you can't turn it off, I don't think) so it should have minimal effect. You can see the args the spikeinterface `run_sorter` takes [here](https://github.com/SpikeInterface/spikeinterface/blob/main/src/spikeinterface/sorters/external/kilosort4.py) and I think it also accepts the kilosort [settings](https://github.com/MouseLand/Kilosort/blob/main/kilosort/parameters.py) arguments here.

In theory the pipelines should be very similar. It you don't turn it off, I'm not sure what will happen,  TBH I don't think it could have a huge affect but might contribute to differences. As Chris mentioned, you need to be aware that if you pass the spikeinterface recording to the sorting analyzer along with the sorter output, the sorting will have been run on motion-corrected data but the recording you are using for quality metrics will not have been.

Are you running everything though spikeinterface, or the raw data directly through kilosort when running kilosort only? If the latter, it will be worth checking the kilosort versions are the same. In both cases, is the data being written to disk?

Alternatively, you can do all preprocessing in spikeinterface. In spikeinterface, the recommendation is to do whitening after motion correction (opposite to kilosort). I think this is because testing on the SI side shows that most of the algorithms (e.g. DREDGE) perform better before whitening. I am not sure how the `kilosort_like` preprocessing method on the spikeinterface side performs after whitening, maybe @samuelgarcia has some insight. 

If you wanted to do all preprocessing in the spikeinterface side, you can do:

1. phase shift
2. bandpass filer
3. common median reference
4. motion correction (e.g. DREDGE, or "kilosort_like") (if you wanted to perfectly match kilosort, you could do whitening before)
5. whiten
6. pass to kilosort with run sorter arguments `skip_ks_preprocessing=False`, `do_correction=False` which I think will turn everything off on the kilosort side. In theory, if you use "kilosort_like" the results even here should be similar to using kilosort4 natively, but worth checking carefully. I am not if there are assumptions buried within the kilosort sorting pipeline that reflect their expected preprocessing chain.

A benefit of this approach if that you have flexibility to use different motion correction methods. You can also then pass the motion-corrected recording to the sorting analyzer, which removes the concern above. 

But before checking with / without motion correction I think it is worth understanding why even just with bandpass filter and common median reference the results are so different. It would be worth trying only on the spikeinterface side:

1. highpass filter (300 Hz)
2. common average reference
3. pass to kilosort with filter and referencing turned off as above.

Because the results should be very similar in this case, and if not we can do some more investigating!

and to summarise the questions:
1) is everything run through spikeinterface so the versions are definately the same?
2) does the problem persist when the CAR and bandpass are turned off in kilosort?
3) are you using a recent version of spikeinterface that saves the preprocessing as binary file? (older ones by default use the spikeinterface object)

darianyao, 2025-06-28:
Thank you for your detailed response. I'm using the demo data provided by kilosort4's official documentation: [(https://kilosort.readthedocs.io/en/latest/tutorials/basic_example.html)]
My code implements two different paths:

First using spikeinterface to do high-pass filter (300Hz cutoff) and CMR preprocessing, then passing the preprocessed recording to the sorter (using kilosort4 with all default parameters), which resulted in 90+ clusters.
Directly feeding raw data to kilosort4, which resulted in 260+ clusters.

This result difference shocked me - why would just doing two redundant preprocessing steps have such a huge impact on the results?
Regarding your questions: I'm using the latest version of spikeinterface, and the preprocessing is saved as binary files. I tried to avoid redundant preprocessing by turning off the relevant kilosort4 preprocessing in the sorter, but found that this resulted in fewer sorted clusters.
Below is the code where I initially discovered the difference.

`# preprocess
recording_filtered = sp.HighpassFilterRecording(recording=recording_loaded,
                        freq_min=300,
                        margin_ms=5)
print(recording_filtered)
recording_cmr = sp.CommonReferenceRecording(recording=recording_filtered,
                        reference='global',
                        operator='median')
print(recording_cmr)
# save preprocessed data
save_preprocessed_to_path = Path('/data/to/preprocessed')
recording_preprocessed = recording_cmr.save(
                        format='binary',
                        name='preprocessed',
                        folder=save_preprocessed_to_path,
                        dtype='int16',
                        n_jobs=4,
                        overwrite=True)
print(recording_preprocessed)`
`# load preprocessed data
datappc_to_path = Path('/data1/account/yaoyipeng/SpinalCord_elecphysiology/Results/DemoData4Kilosort/LoadedData')
recording_preprocessed = si.load_extractor(file_or_folder_or_dict=datappc_to_path)
pprint(recording_preprocessed)

# Here, we're going to use kilosort4 as sorters and get its params
print('kilosort4 params:')
pprint(ss.get_default_sorter_params('kilosort4'))

#  run a sorter
output_to_path     = '/data1/to/SortedTest_no_kl4_preprocessing' 
sorting_ks4 = ss.run_sorter(sorter_name='kilosort4',
                            recording=recording_preprocessed,
                            fs=fs,
                            batch_size=batch_size,
                            nt=nt,
                            bad_channels=bad_channels,
                            output_folder=output_to_path,
                            remove_existing_folder=True,
                            do_CAR=False,
                            highpass_cutoff=300,
                            skip_kilosort_preprocessing=False) #artifact_threshold=artifact_threshold,
print(sorting_ks4)`

Currently, I'm adopting the approach of directly sorting unpreprocessed data, which gives results consistent with using kilosort4 natively. However, I've encountered a new problem: when I use spikeinterface to mimic kilosort4's preprocessing, the SNR is very low (between 0-1), and I'm not sure what's causing this issue. I'm currently working on resolving this. Could you guys please give me some advice for this problem? Thanks again!
`#%% create sortingAnalyzer
analyzer = si.create_sorting_analyzer(sorting=sorting_rm, 
                                      recording=rec_filtered_chanCleaned_cared_hsFiltered,
                                      n_jobs=n_jobs)
print('all units sorted by kilosort4:')
pprint(analyzer)

analyzer_good = analyzer.select_units(unit_ids=good_unit_ids)
print('good units labeled by kilosort4:')
pprint(analyzer_good)

#%% calculate extensions for some metrics
extensions_to_calculate = ['random_spikes', 
                           'waveforms',
                           'principal_components',
                           'templates',
                           'noise_levels',
                           'amplitude_scalings',
                           'correlograms',
                           'isi_histograms',
                           'spike_amplitudes',
                           'spike_locations',
                           'template_metrics',
                           'template_similarity',
                           'unit_locations']
extensions_params = {'random_spikes':{'method': 'uniform', 'max_spikes_per_unit':1000},
                     'waveforms':{'ms_before':1, 'ms_after':2},
                     'principal_components':{'n_components':3, 'mode':'by_channel_local', 'whiten':True},
                     'templates':{'operators':['average']},
                     'correlograms':{'window_ms':50, 'bin_ms':1, 'method':'auto'},
                     'isi_histograms':{'window_ms':50, 'bin_ms':1, 'method':'auto'},
                     'spike_amplitudes':{'peak_sign':'both'},
                     'spike_locations':{'ms_before':1, 'ms_after':2, },
                     'template_metrics':{'include_multi_channel_metrics':True},
                     'template_similarity':{'method':'cosine_similarity'}}
analyzer_good.compute(extensions_to_calculate,
                 extension_params=extensions_params,
                 n_jobs=n_jobs)
print('calculated extensions')
pprint(analyzer_good.get_loaded_extension_names())

# mannually check
amp_per_units = analyzer_good.get_extension('spike_amplitudes')
print('amplitudes per units:', amp_per_units.get_data())
noi_per_units = analyzer_good.get_extension('noise_levels')
print('noise level per units:', noi_per_units.get_data())`

darianyao, 2025-06-28:
Thank you for your detailed response. I'm using the demo data provided by kilosort4's official documentation: [(https://kilosort.readthedocs.io/en/latest/tutorials/basic_example.html)].

My code implements two different paths:
1. First using spikeinterface to do high-pass filter (300Hz cutoff) and CMR preprocessing, then passing the preprocessed recording to the sorter (using kilosort4 with all default parameters), which resulted in 90+ clusters.
2. Directly feeding raw data to kilosort4, which resulted in 260+ clusters.

This result difference shocked me - why would just doing two redundant preprocessing steps have such a huge impact on the results?

Regarding your questions: I'm using the latest version of spikeinterface, and the preprocessing is saved as binary files. I tried to avoid redundant preprocessing by turning off the relevant kilosort4 preprocessing in the sorter, but found that this resulted in fewer sorted clusters.

Below is the code where I initially discovered the difference.

`#load raw data
file_to_path  = '/ZFM-02370_mini.imec0.ap.short.bin'
num_channels  = 384
sampling_freq = 30000
dtype         = 'int16'
recording_npx = si.read_binary(file_paths=file_to_path, 
               sampling_frequency=sampling_freq, 
               num_channels=num_channels,
               dtype=dtype)

#preprocess
recording_filtered = sp.HighpassFilterRecording(recording=recording_loaded,
freq_min=300,
margin_ms=5)
print(recording_filtered)
recording_cmr = sp.CommonReferenceRecording(recording=recording_filtered,
reference='global',
operator='median')
print(recording_cmr)

#save preprocessed data
save_preprocessed_to_path = Path('/path/to/preprocessed data')
recording_preprocessed = recording_cmr.save(
format='binary',
name='preprocessed',
folder=save_preprocessed_to_path,
dtype='int16',
n_jobs=4,
overwrite=True)
print(recording_preprocessed)

 #load preprocessed data
datappc_to_path = Path('/path/to/load/data')
recording_preprocessed = si.load_extractor(file_or_folder_or_dict=datappc_to_path)
pprint(recording_preprocessed)
#Here, we're going to use kilosort4 as sorters and get its params
print('kilosort4 params:')
pprint(ss.get_default_sorter_params('kilosort4'))

#run a sorter
output_to_path = '/path/to/SortedTest_no_kl4_preprocessing'
sorting_ks4 = ss.run_sorter(sorter_name='kilosort4',
recording=recording_preprocessed,
fs=fs,
batch_size=batch_size,
nt=nt,
bad_channels=bad_channels,
output_folder=output_to_path,
remove_existing_folder=True,
do_CAR=False,
highpass_cutoff=300,
skip_kilosort_preprocessing=False) #artifact_threshold=artifact_threshold,
print(sorting_ks4)`

Currently, I'm adopting the approach of directly sorting unpreprocessed data, which gives results consistent with using kilosort4 natively. However, I've encountered a new problem: when I use spikeinterface to mimic kilosort4's preprocessing, the SNR is very low (between 0-1), and I'm not sure what's causing this issue. 

I'm currently working on resolving this. Below is my code.
`#create sortingAnalyzer
analyzer = si.create_sorting_analyzer(sorting=sorting_rm,
recording=rec_filtered_chanCleaned_cared_hsFiltered,
n_jobs=n_jobs)
print('all units sorted by kilosort4:')
pprint(analyzer)
analyzer_good = analyzer.select_units(unit_ids=good_unit_ids)
print('good units labeled by kilosort4:')
pprint(analyzer_good)

#calculate extensions for some metrics
extensions_to_calculate = ['random_spikes',
'waveforms',
'principal_components',
'templates',
'noise_levels',
'amplitude_scalings',
'correlograms',
'isi_histograms',
'spike_amplitudes',
'spike_locations',
'template_metrics',
'template_similarity',
'unit_locations']
extensions_params = {'random_spikes':{'method': 'uniform', 'max_spikes_per_unit':1000},
'waveforms':{'ms_before':1, 'ms_after':2},
'principal_components':{'n_components':3, 'mode':'by_channel_local', 'whiten':True},
'templates':{'operators':['average']},
'correlograms':{'window_ms':50, 'bin_ms':1, 'method':'auto'},
'isi_histograms':{'window_ms':50, 'bin_ms':1, 'method':'auto'},
'spike_amplitudes':{'peak_sign':'both'},
'spike_locations':{'ms_before':1, 'ms_after':2, },
'template_metrics':{'include_multi_channel_metrics':True},
'template_similarity':{'method':'cosine_similarity'}}
analyzer_good.compute(extensions_to_calculate,
extension_params=extensions_params,
n_jobs=n_jobs)
print('calculated extensions')
pprint(analyzer_good.get_loaded_extension_names())

#mannually check
amp_per_units = analyzer_good.get_extension('spike_amplitudes')
print('amplitudes per units:', amp_per_units.get_data())
noi_per_units = analyzer_good.get_extension('noise_levels')
print('noise level per units:', noi_per_units.get_data())`

Could you guys please give me some advice to solve this problem? Thanks again!