SpikeInterface/spikeinterface #4011: Analyzers made since #3794 cannot be opened with older si versions

chrishalcrow, 2025-06-24:
In #3794, we changed `return_scaled` to `return_in_uv`. This includes in the analyzer's `settings.json`. So any analyzer made after #3794 no longer has a return_scaled setting.

Before #3794, the function `load_from_binary_folder` does a `return_scaled = settings["return_scaled"]`. Since the `settings.json` doesn't have return_scaled, this crashes.

Hence, once it comes out, if an analyzer is made using version 103 it won't be loadable by 102. 

Maybe this was known before we merged, but I think it might be too much deprecation all at once? Thoughts?

alejoe91, 2025-06-24:
we definitely need a back-compatibillity fix! @chrishalcrow wanna give it a go?

h-mayorquin, 2025-06-25:
Why do we want to support backporting? isn't it easier to just update?

h-mayorquin, 2025-06-25:
[is an earnest question, not rhetorical]

alejoe91, 2025-06-25:
Ah I misunderstood then!! I agree with @h-mayorquin that in that case upgrading is the simplest option, so IMO it's ok that a new sorting analyzer cannot be opened with an old SI version