SpikeInterface/spikeinterface #4014: Migrating to zarr v3

JulienBrn, 2025-06-26:
Currently spike interface uses zarr v2 which does not have a compatible API with zarr v3 which has different features (mainly sharding). 
My motivation behind sharding is that I like being able to load a single channel pretty fast, so I usually put channel_chunk_size fairly low, but by doing so, I get a huge number of files (384*2*3600 for a 2h recording...).

I've tried to migrate the code of zarrextractors.py and here is my conclusion:

- migrating to v3 is mostly easy by changing `group.create_dataset(key=name,..., data=data)` by `group.create_array(key=name, ..., shape=data.shape, dtype=data.dtype)` followed by `group[name][:] = data`
- we have a problem for structured numpy arrays which were handled in v2 and not in v3. These seem in the process of being handled by zarr, see [this pr](https://github.com/zarr-developers/zarr-python/pull/2874). Either we wish to rely on this future implementation or we can have a custom convention for handling them. A solution could be to create a group with a name like "_structuredarraygrp[name]" and put each individual array in it. 
- adding sharding is extremely easy as its just a parameter of create_array. However, one needs to modify the functions that process arguments.

Is there any interest in such a migration (enough for me to submit a pr) ?

alejoe91, 2025-06-27:
Hi @JulienBrn 

Thank you for this! Definitely there is interest in this. For the structured numpy array, I guess it's only used by the Sorting object, so we could handle it internally as we do now. Generally, the sorting objects can be easily loaded in memory so it shouldn't be a problem. Ok to add sharding args to the write functions :)

Please go ahead and make I PR! We'd be happy to review it and collaborate on it

d-v-b, 2025-06-30:
FYI zarr v3 will support structured dtypes in version 3.1.0, which should be released this week

grahamfindlay, 2025-12-05:
+1, just stumbled into this, wanting to experiment with sharding and, more importantly, use [ome-zarr-python](https://github.com/ome/ome-zarr-py) in the same environment as `spikeinterface`.

closed 2026-10-26 by alejoe91