SpikeInterface/spikeinterface #4016: IBL tests failing

zm711, 2025-06-30:
Seems we have a problem with python 3.13 for the ibl tests from One. We had this previously with a version issue. Maybe @oliche you could comment on what has changed recently?
```
=================================== FAILURES ===================================
______________ TestIblSortingExtractor.test_ibl_sorting_extractor ______________

self = <test_iblextractors.TestIblSortingExtractor testMethod=test_ibl_sorting_extractor>

    def test_ibl_sorting_extractor(self):
        """
        Here we generate spike train with 3 clusters in a very basic ALF format\
        and read it with the spikeinterface extractor
        """
        from one.api import ONE
    
        try:
            one = ONE(
                base_url="https://openalyx.internationalbrainlab.org/",
                ***,
                silent=True,
                cache_dir=None,
            )
        except:
            pytest.skip("Skipping test due to server being down.")
        sorting = read_ibl_sorting(pid=PID, one=one, revision="2023-12-05")
>       assert len(sorting.unit_ids) == 733
E       assert 1091 == 733
E        +  where 1091 = len(array([   0,    1,    2, ..., 1088, 1089, 1090], shape=(1091,)))
E        +    where array([   0,    1,    2, ..., 1088, 1089, 1090], shape=(1091,)) = IblSortingExtractor: 1091 units - 1 segments - 30.0kHz.unit_ids

src/spikeinterface/extractors/tests/test_iblextractors.py:205: AssertionError
```

alejoe91, 2025-06-30:
Tests seem to be failing with `ibllib==3.4.0`. This is a bit strange because the `revision="2023-12-05"` argument that we use in the tests [here](https://github.com/SpikeInterface/spikeinterface/blob/main/src/spikeinterface/extractors/tests/test_iblextractors.py#L204-L207) should ensure we always download the same spike sorting oobject. @oliche any guidance?

h-mayorquin, 2025-06-30:
Should we pin down temporarily back to 3.3.1?

h-mayorquin, 2025-07-01:
Yes, it seems that fixing this to previous version works so the revision is not working as expected?

oliche, 2025-07-01:
Ok, looking into this now