SpikeInterface/spikeinterface #4032: legacy `output_folder` argument name error

ttngu207, 2025-07-04:
There is a minor bug in still having the older `output_folder` argument name in the `run_sorter_container` code - [here](https://github.com/SpikeInterface/spikeinterface/blob/598038ce71dd55e2d854f2ddfe59c44280aa90c4/src/spikeinterface/sorters/runsorter.py#L454)

I believe simply change to `folder=output_folder` on that line would fix the bug.

Below is the error stack caused by the bug (I was running kilosort 2.5 in docker container)

```
ValueError: Invalid parameters: ['output_folder']
Valid parameters are: dict_keys(['detect_threshold', 'projection_threshold', 'preclust_threshold', 'whiteningRange', 'momentum', 'car', 'minFR', 'minfr_goodchannels', 'nblocks', 'sig', 'freq_min', 'sigmaMask', 'lam', 'nPCs', 'ntbuff', 'nfilt_factor', 'NT', 'AUCsplit', 'do_correction', 'wave_length', 'keep_good_only', 'skip_kilosort_preprocessing', 'scaleproc', 'save_rez_to_mat', 'delete_tmp_files', 'delete_recording_dat', 'pool_engine', 'n_jobs', 'chunk_duration', 'progress_bar', 'mp_context', 'max_threads_per_worker'])
```

zm711, 2025-07-05:
Nice catch. Want to try a PR to fix it? :)

ttngu207, 2025-07-05:
Sure! [PR ](https://github.com/SpikeInterface/spikeinterface/pull/4034) issued

closed 2025-07-09 by samuelgarcia