SpikeInterface/spikeinterface #4038: parallel processing with silence_periods

JeffreyBoucher, 2025-07-07:
Hello,

In order to guard against a few periods of time in which some recordings of mine were saturated, I've:

- written and applied a function to detect these periods
- applied silence_periods with the "noise" function to them and the surrounding milliseconds.
- noted these periods for later exclusion from analysis

This process works quite well. However: silence_periods is quite slow when I don't run it parallel. The outer layer of this function does not appear to take an n_jobs kwarg; when I go into silence_periods.py myself and add the kwarg to get_noise_levels, it works, but the parallel pool doesn't properly close afterward, which leads to bugs later when I try to parallel process something else...

So, this post is either a feature request or a question as to how to do this properly if I am not. Is there a way to insist on parallel processing for silence_periods? If not, could you please implement this for future releases? Additionally: is there a spikeinterface-internal way I should be closing the parallel pool after I start one like this?

Thanks!

Jeff Boucher

samuelgarcia, 2025-07-09:
Hi Jeffrey.
I am not sure to understand the question.
This `silence_periods` return a lazy recording that compute on the fly the masking with noise around a trigger list.
In short the get_trace will trigger the "computation" like any other recording.

The parralell will happen if you do a `recording.save(..., n_jobs=10)` or detect_peak on it or any other operation that scan and read by chunk the entire recording.

zm711, 2025-07-09:
I will say getting noise levels is not a lazy process as it requires running to the actual data. So this does break the idea of pure laziness. I don't think this is being applied eagerly to the recording itself at this stage (ie the recording object is still lazy), but getting the noise chunks is not. 

> Spoke a bit too soon I guess, oddly enough when I run detect_peaks just after all this I can neither set an n_jobs manually nor use a global argument... The problem specifically is with noise_level again, internal to this. I don't really understand this, but this used to be the stage where I had problems with my parallel pool not closing... I'll let you know if I can finagle something.

yeah definitely let us know. Also could you let us know your OS and whether you're doing cluster (files + cpus remote) vs accessing data on a network (ie just files are remote analysis local)?

JeffreyBoucher, 2025-07-09:
Getting things like this sometimes too:

[image]

JeffreyBoucher, 2025-07-09:
Alright, getting back into the SI code internals a bit...

I don't know the circumstances in which si.set_global_job_kwargs does or does not apply, but in the peak_detection code (within check_params), eventually this line is reached here, involving get_noise_levels again:

[image]

The random_chunk_kwargs are empty for me. Earlier in the code job_kwargs and method_kwargs are split:

[image]

The method_kwargs do transfer down to check_params, but only the detect_threshold argument (for me) is kept. 

I *think* I might be able to hack a solution here in a few different ways, each of which might break the pool for me again but I plan to start by either canceling the split or by re-adding n_jobs to method_kwargs... It seems odd to me that si.set_global_job_kwargs wouldn't work here; do you know why that might be?

Thanks for your help

JeffreyBoucher, 2025-07-09:
Dug a bit deeper... The first time that fix_job_kwargs is called within this function, it properly extracts the value I set (15). Every other time it extracts some different globals which includes n_jobs of 1. I'll try to figure out when exactly this shift occurs...

zm711, 2025-07-09:
Thanks I think something good be getting messed up with global variable. I haven't tried `silence_periods` myself to know why that would be break things. Could you at the end of a script try doing get_global_job_kwargs? So we could see if we are incorrectly grabbing the kwarg or if we are actually overwriting the kwarg at some point?

JeffreyBoucher, 2025-07-10:
Hello!

At each stage in this I was actually running get_global_job_kwargs, sorry if that wasn't clear; the point it switches is that results = super() bit (or internal to it I guess)

JeffreyBoucher, 2025-07-15:
[spikesort_concatenated.txt](https://github.com/user-attachments/files/21232405/spikesort_concatenated.txt)

Hello,

I was able to contain the problem in a single .py file (changed to .txt so I could place it here). Requirements are pathlib, numpy 1.26.4, spikeinterface 102.3. 

Let me know if you are able to replicate!

Thanks,

Jeff Boucher

zm711, 2025-07-15:
Sweet I'll work on testing that. I'll start by testing the exact same script from main. We will be doing a release soon (ish) so if this is fixed in main I think the advice would be to either install from main or wait for the next release. I don't think we've played with this too much so I assume the bug will still be there.

zm711, 2025-07-15:
I figured out the most immediate issue (on main). For @samuelgarcia we are splitting the `method_kwargs` and `job_kwargs` at the beginning of the node pipeline, but only submitting the `method_kwargs` to the first node of the pipeline. Because we are submitting empty kwargs rather than no kwargs we default back to our defaults rather than the global set by the user. Is there a reason why we don't allow all nodes have access to the job_kwargs? 

Visually,
https://github.com/SpikeInterface/spikeinterface/blob/6bbe42d3a3d439c92d13781c96bdc71586df43a4/src/spikeinterface/sortingcomponents/peak_detection.py#L109-L116

`method_class` is to get the noise levels, but no `job_kwargs`

https://github.com/SpikeInterface/spikeinterface/blob/6bbe42d3a3d439c92d13781c96bdc71586df43a4/src/spikeinterface/core/recording_tools.py#L762C8-L776

in the noise we split the kwargs but because we don't give `job_kwargs` the split leads to an empty dict which is passed on. This empty dict overrides the global_job_kwargs set.

zm711, 2025-07-16:
@yger Sam said you should look at this since we need to see if this is occurring at the level of the node_pipeline or at the level of the `silence_periods`.

samuelgarcia, 2025-07-16:
I think this is because the recording pass to the worker do not contain minimal  propeties like the noise levels.
We should have options for this.
I will check.