SpikeInterface/spikeinterface #4039: about reproduce waveforms

bxy666666, 2025-07-08:
Hi, I want to reproduce unit waveforms , I have had my sorting results, what I want to do is to get the data for plot_unit_waveforms, and let other people plot the waveforms without spikeinterface.

chrishalcrow, 2025-07-08:
Hi @bxy666666,
Confusingly, the `plot_unit_waveforms` actually plots the templates (the averages of the waveforms). So the data you need is in the templates extension. If you have a sorting analyzer an extension computed, you can get the data from any extension by first getting the extension, then getting the data from it. For templates, the syntax is:

```
waveform_data = sorting_analyzer.get_extension("templates").get_data()
```

This returns an array with dimensions (number of units)x(number time samples of template)x(number of channels).

Hope that helps!

bxy666666, 2025-07-08:
thanks, I will try to export the data.

zm711, 2025-07-17:
@bxy666666 I think this should solve the issue so I'll close this. Feel free to reopen or open a new issue if something else comes up.

closed 2025-07-17 by zm711