SpikeInterface/spikeinterface #4048: SpikeGLX Extractor errors out if imec was disabled or not recorded, when using a one box

manimoh, 2025-07-10:
When I try to read a SpikeGLX recording session, where I was only using the Onebox Analog and digital channels; and had disabled the imec (no neurpopixels probes); the code errors out.

```
import spikeinterface.extractors as se
data_path = '/pathtoOneBoxSession'
onebox_rec = se.read_spikeglx(data_path, stream_id='obx0')

---------------------------------------------------------------------------
AssertionError                            Traceback (most recent call last)
Cell In[6], line 1
----> 1 se.read_spikeglx(data_path, stream_id='obx0')

File c:\Users\mvdmlab\miniconda3\envs\pynapple\Lib\site-packages\spikeinterface\extractors\neoextractors\spikeglx.py:82, in SpikeGLXRecordingExtractor.__init__(self, folder_path, load_sync_channel, stream_id, stream_name, all_annotations, use_names_as_ids)
     80 if "lf" in self.stream_id:
     81     meta_filename = meta_filename.replace(".lf", ".ap")
---> 82 probe = probeinterface.read_spikeglx(meta_filename)
     84 if probe.shank_ids is not None:
     85     self.set_probe(probe, in_place=True, group_mode="by_shank")

File c:\Users\mvdmlab\miniconda3\envs\pynapple\Lib\site-packages\probeinterface\neuropixels_tools.py:771, in read_spikeglx(file)
    767 assert meta_file.suffix == ".meta", "'meta_file' should point to the .meta SpikeGLX file"
    769 meta = parse_spikeglx_meta(meta_file)
--> 771 assert "imroTbl" in meta, "Could not find imroTbl field in meta file!"
    772 imro_table = meta["imroTbl"]
    774 # read serial number

AssertionError: Could not find imroTbl field in meta file!

```

I am using the spikeinterface and neo versions installed from the latest version of the main branch.

zm711, 2025-07-10:
I'm not sure if people realized that you could shut recording the probe off. @h-mayorquin seems like you'd be the most interested in this.

h-mayorquin, 2025-07-10:
We don't really have a dedicate space so any place where you can share works for us. How large is the data?