SpikeInterface/spikeinterface #4073: Docker image returning an error for KS2, KS2_5, KS3, KS4, and IronClust

SebastianMarten, 2025-07-18:
Hi, 
I am a very new user of Spikeinterface (si) and am having trouble with using the docker_image option. I would like to run a 30 min segment with different sorters, thus si would be perfect in my case. I am using Python 3.13.,  Spikeinterface 0.102.4 and Docker Desktop on a Windows PC. 
When I try to run a sorter using for example:

`si.run_sorter(sorter_name='kilosort2', recording=rec, folder = r'D:\GL1060_20210817_sleep4_g0_imec0\data\kilosort2_output', docker_image=True, remove_existing_folder=True, with_output=True)`

I get the error:

zm711, 2025-07-18:
Sorry we are little behind on updating our containers. @alejoe91 can double check to see if this is a stale container issue or a script issue when he gets a moment.

SebastianMarten, 2025-07-18:
Hi @alejoe91,

Thanks a lot for the quick response! May I ask do you think I will be able to get those running locally or am I bound to KS4 with this GPU now? 

Thanks a lot for the quick response, much appreciated!

zm711, 2025-07-18:
> May I ask whether you think I will be able to get those running locally or am I bound to KS4 with this GPU now?

Could you explain this question a bit more? Are you asking if you could install the other sorters locally ?

SebastianMarten, 2025-07-18:
Hi @zm711, 

I am trying to get KS 2, KS 2.5 or KS 4 to run but I am experiencing some issues with all of them. Even when attempting to install them locally I run into problems. From what is understand these issues arise since the graphics card or its required software (e.g., NVIDIA CUDA toolkit) is not compatible with for example Matlab. Thus, my question, whether I should consider downgrading to ensure compatibility. I was hoping you could give me advise on that.

Thanks a lot!

zm711, 2025-07-18:
Yeah getting KS2 and KS2.5 to work via Matlab is definitely a pain. One issue is that some versions of Matlab often error during running KS because of a change in one of the functions (I believe the change occurred around 2021/2022)--Aurélien tried to submit a PR to fix the issue but they closed it. So in our group people always use Matlab < 2022. Then you do have to make sure your GPU can work with older Matlab and find a CUDA toolkit that fits between them which is a struggle. For KS4 last I checked it worked pretty well to just follow their repo instructions so I guess some further questions would be:

Confirming you're on Windows (and which version)?
Which GPU do you have (Alessio is good at seeing that info but for me it is easier if it just listed)?
And what errors are you actually getting preventing local install?

SebastianMarten, 2025-07-21:
I am working with Windows 11 Enterprise 24H2 and an NVIDIA RTX 5070 Ti. Kilosort4 worked very well with the standard repo installation. I also got Kilosort 2.5 to run with the following versions and ` parallel.gpu.enableCUDAForwardCompatibility(true)` :

- Microsoft Visual Studio Enterprise 2019 (Desktop development with C++ toolkit)
- NVIDIA CUDA Toolkit 11.0
- MATLAB 2021b

I think I will work with this for now.
Thanks a lot for your quick responses!

mandarmp, 2025-09-22:
I need some advice. So If I buy a 5090 then I can locally install KS4 and integrate with the spikeinterface pipeline? Am i correct?

mandarmp, 2025-09-27:
Thanks @alejoe91 !

alejoe91, 2025-09-28:
Closing this since all images have been updated

closed 2025-09-28 by alejoe91

SebastianMarten, 2025-11-05:
@mandarmp  Oh I see. I am talking outside the docker with a local installation.