SpikeInterface/spikeinterface #4081: Dataset not availabe

Mahdi-Nouradini, 2025-07-24:
How can I access the `cambridge_data.dat` dataset. 
The drive link is not found on the server.

zm711, 2025-07-24:
could you share a link to where you are trying to get this file? I'm not sure which dataset it is that you want?

zm711, 2025-07-24:
I just tried and the link brought me right to the file. could you print the exact error you are getting?

Mahdi-Nouradini, 2025-07-24:
I get this error:

"404. That’s an error.

The requested URL was not found on this server. That’s all we know."

zm711, 2025-07-24:
Also we have an api level change for >= 0.101.0 so I would recommend if you are starting out to try out more recent tutorials :) Just so you learn the `SortingAnalyzer` instead of the `WaveformExtractor`.

Mahdi-Nouradini, 2025-07-24:
Of course.
Your link is ok. 
Thank you

zm711, 2025-07-24:
Sweet. I'll close this for now, but if you run into any more issues feel free to open a new issue. We are always happy to help!

Mahdi-Nouradini, 2025-07-24:
Actually I am new to spikeinterface. 
I have some questions, I really appreciate you if you can help me ...

Mahdi-Nouradini, 2025-07-24:
Thank you so much. 
I want to save and analyze data in the PCA feature space. Is there any way to extract data from spikeinterface?

zm711, 2025-07-24:
Yes the idea would be to take a `SortingAnalyzer` and then run some version of 

```python
import spikeinterface.full as si
analyzer = si.create_sorting_analyzer(sorting=sorting, recording=recording)
analyzer.compute(['random_spikes', 'waveforms', 'templates', 'principal_components'])
pcas = analyzer.get_extension('principal_components').get_data()
```
Then you'll have PCA from the waveforms of the sorting.

Note that a good pipeline will likely need to preprocess the recording a bit (`cmr` and `filtering` are common) so depending on how much of a beginner you are you may want to get a deeper understanding with some of these tutorials and how tos below:

I would check out this tutorial for understanding how the analyzer works.
https://spikeinterface.readthedocs.io/en/stable/tutorials/core/plot_4_sorting_analyzer.html#sphx-glr-tutorials-core-plot-4-sorting-analyzer-py

This tutorial if you want a deep dive on creating a whole pipeline (for neuropixel, but can be adapted to any probe)
https://spikeinterface.readthedocs.io/en/stable/how_to/analyze_neuropixels.html

this describes the `principal_components` extension
https://spikeinterface.readthedocs.io/en/stable/modules/postprocessing.html

and if you're comfortable reading code the actual PCA code is here:
https://github.com/SpikeInterface/spikeinterface/blob/main/src/spikeinterface/postprocessing/principal_component.py

Mahdi-Nouradini, 2025-07-25:
I have extracted PCA features.
My question is:
In such a case, where the PCA features appear tightly clustered with no obvious separation, what does the clustering stage (e.g., using Mountainsort or any other sorter) actually rely on to distinguish between units?

Is the clustering still meaningful even when PCA looks like a blob? Or is it possible that the feature extraction or waveform alignment is not set up correctly?

Any insight into how sorting algorithms interpret such feature spaces would be greatly appreciated!

Thanks in advance.

zm711, 2025-07-25:
Hey @Mahdi-Nouradini that's a huge question! Honestly I would probably watch this tutorial by Pierre to better understand the steps of spike sorting (including clustering)

https://www.youtube.com/watch?v=ydJypW3tTl0

It's one hour long, but it really goes step by step how spike sorting works. each spike sorter will do other tricks around the edges, but Pierre does a great job of explaining the key steps for all modern sorters.

Finally for moutainsort they use their isosplit algorithm which I don't think uses PCA but I could be wrong. They have a paper on which specific technique they use.

Mahdi-Nouradini, 2025-07-25:
Thank you for your help. I watched the tutorial by Pierre and mostly understood the steps of spike sorting. 
Do you know that how can I access the slides of the tutorial ?