SpikeInterface/spikeinterface #4083: "read_openephys"with onebox ADC stream Error

jialeren2024, 2025-07-24:
Spikeinterface  0.102.3. failed to load the timestamps use "read_openephys" with the OneBox-ADC stream, a Error occurred, that didn't happen in the old version: might be due to a mismatch between the stream name: 'OneBox-ADC_ADC' and the folder name:'OneBox-ADC'

[image]
<img width="580" height="99" alt="Image" src="https://github.com/user-attachments/assets/ebd53356-56e4-426b-abe7-467a

a sample data for test:
[2025-06-03_14-42-36.zip](https://github.com/user-attachments/files/21419901/2025-06-03_14-42-36.zip)

6399633c" />

chrishalcrow, 2025-07-27:
Hello, there are two errors going on here:

1. The error in the error trace "cannot access local variable 'warnings' where it is not associated with a value"

This is very weird: there's an import of the warnings package at the top of the file, and that's been there for 3 years... I have no idea what's going on

2. (more important!) You can't open the timestamps.

As you rightly figured out, we currently assume that the `oe_stream` folder (where timestamps are kept) is named "whatever-comes-after-#-in-the-steam-name". In this case, `OneBox-100.OneBox-ADC_ADC`. But the sample data shows that it's equal to `OneBox-100.OneBox-ADC`. To fix this problem generally, we need more info on how OneBox is naming folders. Anyone know who to contact??

alejoe91, 2025-07-28:
The warning issue has been fixed here already: https://github.com/SpikeInterface/spikeinterface/pull/3980

Let's open an issue on the Open Ephys repo

alejoe91, 2025-07-28:
Actually, this is an issue in NEO: https://github.com/NeuralEnsemble/python-neo/blob/master/neo/rawio/openephysbinaryrawio.py#L205-L217

We basically add the `_ADC` to the stream if the channel name has ADC in it. This is incompatible with OneBox, since the ADC has its own stream already.

alejoe91, 2025-07-28:
@jialeren2024 should be fixed by this PR in NEO: https://github.com/NeuralEnsemble/python-neo/pull/1753

Can you try to install neo and spikeinterface from source and test it again?

closed 2025-07-31 by alejoe91