SpikeInterface/spikeinterface #4088: How to “get started” functions incompatible with Python 3.13

Jai477, 2025-07-28:
Hello, I am new to SpikeInterface and I am attempting to test it for my lab's extracellular ephys data. I'm going through the installation and requirements to test the sample data and I'm running into issues trying to open the data to start the analysis process. I will also note I am new the Python so I am learning as I go and I apologize if this has been addressed elsewhere in the Git Hub.

MEArec is conflicting with the numpy version because of elephant 1.1.1. I saw in another issue thread this was addressed, but I'm assuming that was for users who are actively working with their own data in SpikeInterface and would not be specific for the "How to get started" tutorial. #3683 

It seems like python began to run the MEArec module but then another error came up. The python package for that is not compatible with 3.13 it seems. And on the MEArec website it says distutils has been replaced with packaging, as well as this issue thread https://github.com/NeuralEnsemble/python-neo/pull/1135.

I was just wondering if there is any plans to update the how to get started process or if there is another doc I can use to run through set-up procedures? I want to make sure I know the basics before I attempt any complex steps. Thanks for any help that can be offered!

```import MEArec as mr
Traceback (most recent call last):
  File "<python-input-9>", line 1, in <module>
    import MEArec as mr
  File "C:\Users\jaime\AppData\Local\Programs\Python\Python313\Lib\site-packages\MEArec\__init__.py", line 5, in <module>
    from MEArec.generation_tools import (gen_recordings, gen_spiketrains,
                                         gen_templates)
  File "C:\Users\jaime\AppData\Local\Programs\Python\Python313\Lib\site-packages\MEArec\generation_tools.py", line 10, in <module>
    from .generators import (RecordingGenerator, SpikeTrainGenerator,
                             TemplateGenerator)
  File "C:\Users\jaime\AppData\Local\Programs\Python\Python313\Lib\site-packages\MEArec\generators\__init__.py", line 1, in <module>
    from .recordinggenerator import RecordingGenerator
  File "C:\Users\jaime\AppData\Local\Programs\Python\Python313\Lib\site-packages\MEArec\generators\recordinggenerator.py", line 10, in <module>
    from distutils.log import DEBUG
ModuleNotFoundError: No module named 'distutils'```

zm711, 2025-07-28:
Howdy @Jai477,

Thanks for the report. Could you:

1) link to the doc page that is causing the problem. We are slowly moving our docs away from MEArec to help with some of this. 
2) how did you install spikeinterface? (we are recommending uv over conda currently)

Jai477, 2025-07-28:
Hi!
Here's the doc I am following: https://spikeinterface.readthedocs.io/en/0.100.3/how_to/get_started.html

I installed using the pip command in my command prompt window. I am adding my text files for everything I've done so far. The cont files are just the more recent runs where I've run into the MEArec issues, and the other two files are the beginning of downloading SpikeInterface.

[Command Prompt_Spike_Install_Cont.txt](https://github.com/user-attachments/files/21475305/Command.Prompt_Spike_Install_Cont.txt)
[Command Prompt_SpikeInterface_Install.txt](https://github.com/user-attachments/files/21475304/Command.Prompt_SpikeInterface_Install.txt)
[Python 3.13_SpikeInterface_Tutorial1.txt](https://github.com/user-attachments/files/21475306/Python.3.13_SpikeInterface_Tutorial1.txt)
[Python 3.13_SpikeTutorial_Cont.txt](https://github.com/user-attachments/files/21475307/Python.3.13_SpikeTutorial_Cont.txt)

Jai477, 2025-07-28:
I did not save the text file for installing spikeinterface initially but I do know that I installed using this:

`py -3.13 -m pip install spikeinterface[full,widgets]`

zm711, 2025-07-28:
It seems like you are trying to use standalone python on Windows which is a big struggle. Would you be willing to try out `uv`? For python beginners we provide tips here https://github.com/SpikeInterface/spikeinterface/tree/main/installation_tips. They provide info on both `uv` and on `conda`. I see you tried `venv` at one point which is a good idea. It is easy to kill your install trying to mix and match packages so using a conda env or making a venv with uv can be a great way to isolate as you are playing around. 

If you install uv you could do something as simple as 

```
uv venv si_env --python 3.12 # or 3.13
si_env\Scripts\activate
uv pip install spikeinterface[full,widgets]
```

then rather than use MEarec I would recommend trying your own data. So let's say you use intan to record data you could then do
```
python
>>> import spikeinterface.full as si
>>> recording = si.read_intan('path/to/rhd.rhd', stream_id='0')
```
then just follow the tutorial as is. If you really want to do MEARec then above you should do

```
uv venv si_env --python 3.12 # or 3.13
si_env\Scripts\activate
uv pip install spikeinterface[full,widgets,docs]
```

The biggest issue with python is that you need to do dependency resolution. As you are trying to install a bunch of packages 1 or 2 at a time they are getting out of sync and you're destroying the setup. So it's better to do one big round of resolution (either with pip or uv or conda) at once. You were missing the `,docs` which to use MEArec is necessary. We also have issues with people getting `datalad` working so that you would need to do separately. That's why I recommend trying to load your own data.

If you don't have data yet you can also just generate data then follow the tutorial. So instead of `se.read_mearec` you could do

```python
recording, sorting = si.generate_ground_truth_sorting([600], sampling_frequency=30_000) # this makes a tetrode, but if you want more than add in `num_channels=32` or something
````

zm711, 2025-07-28:
Great question! Depends on how you made it. Conda stores all your envs in a specific folder on your computer. So to activate any environment you just type `conda activate xxxx` to activate that env. If you were using `uv` then it creates the venv folder wherever you were. So if you were in your documents folder you would just go back to the documents and type
`si_env/Scripts/activate` and then you should see `(si_env)` appear in the command prompt. That tells you that you successfully activated the isolated environment. 

One little Windows pitfall with python to remember is that because Windows paths use `\` instead of `/` like Linux/mac you can accidentally trigger formatting. So when dealing with Windows paths in python it is usually easier just to use the raw string version which just requires adding `r` to the start of the string.

so, don't do
```
"path\to\my\data"
``` because that will add a "tab" at the `\t` which will break the path
type
```
r"path\to\my\data"
```
which will prevent the formatting. (You can also do escaping with `\\` but that can get messy.

Jai477, 2025-07-28:
Okay, great! Thank you for this breakdown. I was able to find it in my user folder.

Would you recommend changing the si_env path to my documents folder? It is currently saved to my user folder as a default.

Also if I were to do another environment, and if I am in the same general folder (for example my current default user path) would `(si_env)` create another folder or overwrite the other environment?

zm711, 2025-07-28:
No worries. Happy to help. We all have to start somewhere!

I would say open a new issue. This one was based on packaging issues and not with the tutorial itself. So if you run into an issue we can doublecheck the tutorial itself. Or other questions can be opened as issues (we are a repo that doesn't distinguish between issues/questions for the issues section--other repos do).

closed 2025-07-28 by zm711

zm711, 2025-07-28:
Also if you like videos let me recommend our youtube channel where we go through things live!

https://www.youtube.com/@Spikeinterface