SpikeInterface/spikeinterface #4094: CUDA out of memory

yyyaaaaaaa, 2025-07-30:
I was sorting kilosort4 with spikeinterface and had the problem of running out of CUDA memory, but I didn't have the above issue with the kilosort4 GUI.

Error message.
SpikeSortingError: Spike sorting error trace:
Traceback (most recent call last):
  File "D:\Software\spikeinterface-main\spikeinterface-main\src\spikeinterface\sorters\basesorter.py", line 270, in run_from_folder
    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)
  File "D:\Software\spikeinterface-main\spikeinterface-main\src\spikeinterface\sorters\external\kilosort4.py", line 380, in _run_from_folder
    st, tF, _, _ = detect_spikes(**detect_spikes_kwargs)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\Software\Miniconda\envs\ssi_copy\Lib\site-packages\kilosort\run_kilosort.py", line 797, in detect_spikes
    Wall3 = template_matching.postprocess_templates(
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\Software\Miniconda\envs\ssi_copy\Lib\site-packages\kilosort\template_matching.py", line 144, in postprocess_templates
    Wall2, _ = align_U(Wall, ops, device=device)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\Software\Miniconda\envs\ssi_copy\Lib\site-packages\kilosort\template_matching.py", line 132, in align_U
    Xmax = X.abs().max(0)[0].max(0)[0].reshape(-1, ops['nt'])
           ^^^^^^^
torch.OutOfMemoryError: CUDA out of memory. Tried to allocate 41.20 GiB. GPU 0 has a total capacity of 15.92 GiB of which 0 bytes is free. Of the allocated memory 48.08 GiB is allocated by PyTorch, and 5.97 MiB is reserved by PyTorch but unallocated. If reserved but unallocated memory is large try setting PYTORCH_CUDA_ALLOC_CONF=expandable_segments:True to avoid fragmentation.  See documentation for Memory Management  (https://pytorch.org/docs/stable/notes/cuda.html#environment-variables)
Spike sorting failed. You can inspect the runtime trace in C:\data_csy\ks43/spikeinterface_log.json.

I read the raw data and then use the .save() function to save it as a binary file, and then load it as a record file for kilosort4。
recording = recording.save(format='binary',
name='unpreprocessed',
folder=Path_unpreprocessed_binary,
dtype='int16',
n_jobs=8,
overwrite=True)
recording2 = si.load(file_or_folder_or_dict=Path_unpreprocessed_binary)
pprint(recording2)

zm711, 2025-07-30:
Which version of spikeinterface and which version of kilosort? These out of memory errors happened a lot with early version of kilosort so maybe they have added another update to flush more often that we need to review.

zm711, 2025-07-30:
I don't think we have officially tested against 4.1.0. Let me loop @alejoe91 in !

yyyaaaaaaa, 2025-07-30:
> Has the 4.0.39 version of ks been tested, I am going to revert to this version to see if there is a problem. [@zm711](https://github.com/zm711)

I've tried the 4.0.39 version of ks and the same code works fine. However, I still keep the 0.102.4+4.1.0 environment to keep track of your latest progress.

zm711, 2025-07-30:
> Has the 4.0.39 version of ks been tested, I am going to revert to this version to see if there is a problem. @zm711 

Yeah we've tested up that hi. I'm worried a CUDA memory error might be a bit tricky to find--ie I don't think our testing will necessarily pick it up. But we run the KS tests on a cron job so we can start working on updating to see what they changed soon. Thanks for the report!

samuelgarcia, 2025-07-31:
I tested ks 4.1.0 yesterday and get also cuda memory problems because my GPU is "only" 8G.
So I switvh back to 4.0.39 and then it is working again.

yyyaaaaaaa, 2025-08-02:
@zm711 
In Kilosort 4.1.0, compared with version 4.0.39, two default parameters have changed: max_cluster_subset and cluster_downsampling. After reverting these two parameters to the 4.0.39 settings(max_cluster_subset =None,cluster_downsampling=20,The same parameter settings were also applied in the Kilosort 4.1.0 GUI.), the pipeline runs normally.

However, a new issue has emerged: although the Kilosort version is the same, the results obtained from running Kilosort via SpikeInterface are different from those obtained via the GUI, even though both use the same source data.
Specifically, the number of sorted units from SpikeInterface is 236, while the GUI-based run yields 273 units.
I have read issue #4007  and followed the same strategy — completely skipping SpikeInterface's preprocessing and relying solely on Kilosort's built-in preprocessing steps.

The only difference is that my source data is based on MaxOne and in uint16 format. I used:
si.preprocessing.unsigned_to_signed(recording=recording, bit_depth=10)
to convert it to int16 format.
I also tried:
ks.io.spikeinterface_to_binary(dtype=np.int16)
to convert the original MaxOne data into a Kilosort-compatible binary format and then ran Kilosort using the GUI.
Both of these conversions result in the same number of units when using the GUI, but still differ from the SpikeInterface-based Kilosort run.

Here is my SpikeInterface sorter configuration:
sorting_ks4 = ss.run_sorter(
    sorter_name='kilosort4',
    recording=recording2,
    folder=Path_ks4,
    remove_existing_folder=True,
    skip_kilosort_preprocessing=False,
    do_CAR=True,
    highpass_cutoff=300,
    fs=20000,
    batch_size=30000,
    nt=int(0.02*20000+1),
    nblocks=0,
    dminx=35,
    bad_channels=None,
    max_cluster_subset=None,
    cluster_downsampling=20,
)
SpikeInterface version: 0.104
Kilosort version: 4.1.0

b-grimaud, 2025-08-22:
I gave a shot to KS4 for the first time and I can confirm downgrading to 4.0.39 fixes the issue.
I was running 4.1.0 on Ubuntu 24.04, on a RTX A4000 with CUDA 12.4 and 15 GB of vRAM.
Reducing the batch size helped enough to go through spike detection but still crashed at the first clustering, as it was trying to allocate 40+ GB of memory on the GPU. Setting `batch_size` to anything less than ~1500 seems to crash as well.

closed 2025-08-24 by yyyaaaaaaa