SpikeInterface/spikeinterface #4114: error in compare_multiple_sorters

paul-aparicio, 2025-08-22:
Hello, I am using spikeinterface 0.103.0 on ubuntu 20.04.6. I am getting an error when using ``si.compare_multiple_sorters`` 

``ValueError: make_best_match or make_hungarian_match has unit_ids dtype wich are not  'i' or 'U'``

I am not sure what is happening, any help would be very appreciated!

paul-aparicio, 2025-08-22:
sorry if I am doing this wrong (I am more of an experimenter than a coder!), but I was able to find in the code:

``def _empty_match_series(unit1_ids, unit2_ids):
    # construct empty series of match with the correct dtype for best match and hugarian match
    import pandas as pd

    match_12 = pd.Series(data=np.zeros(unit1_ids.size, dtype=unit2_ids.dtype), index=unit1_ids)
    if unit2_ids.dtype.kind == "i":
        match_12[:] = -1
    elif unit2_ids.dtype.kind == "U":
        match_12[:] = ""
    elif unit2_ids.dtype.kind == "O":
        match_12[:] = ""
    else:
        raise ValueError("make_best_match or make_hungarian_match has unit_ids dtype wich are not  'i' or 'U'")
    return match_12
``

And I see that when I run ``print(sorting_IC.unit_ids.dtype.kind)`` I get "u" (lowercase u), but the code is looking for uppercase "U?" Maybe that is why it doesn't get a match and throws the error?

h-mayorquin, 2025-08-22:
Yeah, still + 1 on restricting ids to strings.

Integers are for doing arithmetic.

paul-aparicio, 2025-08-27:
Sorry! Yes. It worked well.