SpikeInterface/spikeinterface #4115: issue with "get_traces"

jialeren2024, 2025-08-22:
The get_traces function doesn't seem to correctly find the channels for ADC signals, return: (numsamples, 0), 
while it works normally for AP and LFP signals. This function uses ids_to_indices to find the channel list, but the indices of the ADC stream looks correctly.

[image]

in baserecoring.py :

`channel_indices = self.ids_to_indices(channel_ids, prefer_slice=True)
        rs = self._recording_segments[segment_index]
        start_frame = int(start_frame) if start_frame is not None else 0
        num_samples = rs.get_num_samples()
        end_frame = int(min(end_frame, num_samples)) if end_frame is not None else num_samples
        traces = rs.get_traces(start_frame=start_frame, end_frame=end_frame, channel_indices=channel_indices)`

zm711, 2025-08-24:
Thanks @jialeren2024! We don't officially support open ephys onebox. We are actively implementing it in python-neo right now so stick with us as the stream organization is slightly different so we need to update the open-ephy code. Feel free to check out the PR [here](https://github.com/NeuralEnsemble/python-neo/pull/1753).

If you're comfortable with GitHub/git you can install from that PR and test if that fixes the issue for you. If it does please post to that PR. We don't have test data on our testing-repo yet so Alessio was doing the initially with I think some private data he had.

jialeren2024, 2025-08-26:
It works, thank you so much!
Fell free to use that data,I'm glad it can help.

zm711, 2025-08-27:
Thanks @jialeren2024 we will get that added on the gin side and then merge at the Neo level. Since you use onebox and openephys definitely report any more issues you run into since we are newly supporting this! Thanks for letting us use the file for testing.